efabless/caravel #6: Housekeeping SRAM read-only access should be expanded to allow full addressing

RTimothyEdwards, 2021-12-29:
Currently the housekeeping module reserves one SPI register (8 bits) for accessing the SRAM read-only port.  The implementation in `caravel_pico` is to map this address to the upper 256 words of the SRAM.  It got this way because historically, the SRAM was two modules of 1kB each (256 words each), and one of those SRAM blocks was given to the housekeeping for read-back, and the other memory-mapped into the main memory space.  The current implementation uses the "csb" bit from housekeeping to toggle the use of the upper half of the SRAM between housekeeping and the SoC.  It would make more sense to give another bit to the address in housekeeping and let housekeeping read the full 512 words.

RTimothyEdwards, 2022-09-28:
This is still warranted, but the current VexRISC implementation does not export the read-only 2nd port of the SRAM, so it is unimplementable without additional development of the VexRISC LiteX core.

azwefabless, 2022-10-04:
Unused port must be correctly terminated.

RTimothyEdwards, 2022-10-05:
FYI, this issue has changed definition from a requested enhancement to the housekeeping (which has not been done and will be deferred) to a fix in the VexRISC LiteX core (which is urgent and must be done for the upcoming tapeout).