efabless/caravel #15: Incorrect bit values for pullup/pulldown functions in defs.h

RTimothyEdwards, 2022-01-19:
As noted by Sylvain Munaut, the bit values for the pullup and pulldown input functions are incorrect in defs.h.
The existing definitions are:
```
#define GPIO_MODE_MGMT_STD_INPUT_NOPULL    0x0403
#define GPIO_MODE_MGMT_STD_INPUT_PULLDOWN  0x0803
#define GPIO_MODE_USER_STD_INPUT_NOPULL    0x0402
#define GPIO_MODE_USER_STD_INPUT_PULLDOWN  0x0802
```

However, bit 1 set to 1 tristates the output buffer and prevents the output from driving the pullup/pulldown resistor.  Instead, these definitions should be:

```
#define GPIO_MODE_MGMT_STD_INPUT_NOPULL    0x0401
#define GPIO_MODE_MGMT_STD_INPUT_PULLDOWN  0x0801
#define GPIO_MODE_USER_STD_INPUT_NOPULL    0x0400
#define GPIO_MODE_USER_STD_INPUT_PULLDOWN  0x0800
```

The `defs.h` file is the only thing that needs to be changed.

smunaut, 2022-01-23:
Small mistake in the above, the `NOPULL` version shouldn't change, it's the `PULLUP` version that also needs change

RTimothyEdwards, 2022-03-22:
@jeffdi :  I just created a pull request for this, noting Sylvain's comment above.

RTimothyEdwards, 2022-09-28:
This error has migrated (unfortunately) to `caravel_mgmt_soc_litex`.  I will open an issue there.

RTimothyEdwards, 2022-09-28:
Note that the error was already corrected in the caravel repository.  I am closing this issue here.  See the new pull request in caravel_mgmt_soc_litex (pull request #47).

closed 2022-09-28 by RTimothyEdwards