efabless/caravel #20: IO pins on user_analog_project_wrapper

adithyasunil26, 2022-01-26:
It would be great to know the structure of the `gpio_analog`, `gpio_noesd` and `io_analog` pins on the pad-frame. Knowing the load resistance or capacitance present would help in designing efficient measurement circuits and buffers for analog designs.

armleo, 2022-02-06:
gpio noesd ~ 200 ohm, gpio_analog ~300 ohm, io analog ~ 100 ohm

RTimothyEdwards, 2022-03-22:
(Somehow I missed this when it was posted)
(@jeffdi ---  We should enable the "discussions" on the repository for these sort of FAQ-style questions)

The best documentation for the I/O connections can be found in the https://github.com/efabless/caravel_user_project_analog repository under `verilog/rtl/user_proj_example.v`.

closed 2022-10-20 by RTimothyEdwards