efabless/caravel #26: SPI Passthrough has incorrect timing for clk signal

westonb, 2022-02-04:
As observed on my MPW2b board (chipignite with metal fix) the SPI passthrough has incorrect timings, which causes sporadic issues with being able to program the SPI flash, especially being unable to erase the flash. 

The SPI interface should be mode 0, which keeps the clock line low between transactions and data is sampled on rising clock edge and shifted out on the falling edge. However, the passthrough interface keeps the clock line high, which adds extra clock transitions. For the added falling edge where the clock line is brought low this is not an issue as the clock line is brought low before the !CS signal for the flash is brought low, so it is ignored. However, when the transaction is ended the !CS line and clock line are brought high at the same time. If the clock line is registered as being brought high first this effectively clocks out a 9th bit. This is dependent on a small difference in timing between the two signals, which varies between ICs and across temperature. On one board I have no issues with programming but on another board I need a small capacitance on the clock line so it goes high slightly later. 

This extra bit most commonly impacts the erase command as the erase command needs to directly follow the write enable command and the extra 9th bit seems to be parsed as an attempted command. 

Here is a scope capture of the transaction on the data lines connecting the SPI flash and caravel when the chip is in SPI passthrough mode. green is clock, orange is !CS, yellow is data, blue is the clock from the FTDI (ignore the ringing, I am missing a ground lead on that one). You can see how the clock line is brought low before the !CS line but they go high at the same time (its a bit hard to see because the traces overlap). 

[image]

RTimothyEdwards, 2022-02-05:
The issue is caused by a pull-up 10k ohm resistor on the flash SPI clock pin on the development board.  Another underlying problem, though, is that the pass-through mode simply transfers control of the flash SPI control pins back to the management SoC.  The management SoC keeps these pins in tristate when reset is asserted.  So to make the processor robust to this issue, the pass-through mode should be modified to always transfer of the CSB pin at least one clock cycle before it transfers control of the CLK pin, so that regardless of any pull-up resistor on the line or reset state of the processor, the clock line will not transition high while the chip is still selected.

RTimothyEdwards, 2022-09-27:
@jeffdi :  This specific issue was resolved, but a similar issue existed with the user pass-through mode and was fixed in the branch `fix_user_pass_thru` which has not been merged, but needs to be (it was committed but a pull request was never made, apparently).

RTimothyEdwards, 2022-09-27:
@jeffdi :  I just made that pull request, which is #134

azwefabless, 2022-10-05:
Design reviewed for implementation but needs verification test.