efabless/caravel #31: simple POR needs to separate the ground domains properly

RTimothyEdwards, 2022-02-16:
As pointed out by Mitch Bailey, the ground domain of the HVL inverters used for level shifting from 3.3V to 1.8V is tied to the high voltage ground domain.  This connection is left over from the original design, which had only one ground connection for the whole circuit.  Even with the current design, since the ground domains are connected through the substrate, it is not strictly an error, but it is good practice not to have standard cells crossing voltage domains.

Okay solution:  Do not put "isosub" under the level shifters, and connect the level shifter ground rail to VSS1V8.

Better solution:  Put the whole POR circuit in a deep nwell and cleanly separate the ground domains.

(Really, going forward, all of the digital circuits that have more than one power domain should be placed in deep nwell structures that keep the ground domains cleanly separated.  That includes the "management protect" block and the "gpio_control_block" cell in addition to the simple_por cell.)

d-m-bailey, 2022-04-19:
@RTimothyEdwards Haven't completed LVS yet, but these are the temporary changes I made to `simple_por` (for reference).
[image]

d-m-bailey, 2022-04-20:
I think the SUBCUT is incorrect. I think it should cover everything except where it is now in `simple_por`.

d-m-bailey, 2022-07-04:
I believe this was corrected with #90 and can be closed. It appears **not** to be integrated into mpw-6.

RTimothyEdwards, 2022-09-27:
Yes, the simple POR GDS is correct in the current caravel version and will be applied to all versions from MPW-seven forward.

closed 2022-09-27 by RTimothyEdwards