efabless/caravel #34: Instance name is using a System Verilog keyword

kauser-rl, 2022-02-22:
I'm trying to simulate the Caravel SoC with our SystemVerilog based design. I'm using `Verilator` with SV enabled and hitting the below issue. Is it possible to rename the instance name please? I'm working around it locally by renaming the instance, but it is blocking me from making the simulation infrastructure available to others without the manual workaround.

`
%Error: ../../caravel_soc/verilog/rtl/caravel.v:614:22: syntax error, unexpected clocking, expecting IDENTIFIER or randomize
  614 |     caravel_clocking clocking(
      |                      ^~~~~~~~
`

kauser-rl, 2022-02-24:
I can reproduce this issue using `Icarus Verilog` too if I use switches to enable newer version of the standard.

mattvenn, 2022-03-10:
I think this is now resolved, I don't see any reference to 'instance' now in the caravel.v verilog.

kauser-rl, 2022-03-10:
I still see it in the `mpw-5c` tag. Are you looking for the instance name `clocking` for the `caravel_clocking` module? It appears to still be there. Maybe it is fixed in HEAD, if so, any plans for a new tag?

RTimothyEdwards, 2022-03-22:
@mattvenn , @jeffdi :  I have just created a pull request for this

RTimothyEdwards, 2022-04-07:
The fix for this has just been merged and will be in effect starting with the tags for MPW-five.

closed 2022-10-20 by RTimothyEdwards