efabless/caravel #40: MPW-5 Tapeout fails with FEOL DRC violations

BrandonOng22, 2022-03-02:
The precheck passed successfully. Tapeout fails with 4 FEOL drc violations. The violations are 'nwell.6' in the H3_sky130_fd_io__gpiov22_amux cell. The violation occurs outside of my design.

`STDERR: [ WARN ] FEOL Check Result: 'caravel_00054489.oas' has 4 DRC violations.`

[image]

mattvenn, 2022-03-10:
thanks, this is a known issue and it's been solved in open_pdks already. Now we are waiting on it going live on the efabless server.

BrandonOng22, 2022-03-10:
> thanks, this is a known issue and it's been solved in open_pdks already. Now we are waiting on it going live on the efabless server.

Thanks Matt. Is this something that I should just wait until it gets updated?

RTimothyEdwards, 2022-03-22:
@mattvenn :  This was solved with a "hack" for MPW-five (failing cells are ignored by klayout DRC) and the issue should remain open until the proper fix (in open_pdks) gets applied to the server;  and at that time we need to make sure that the klayout DRC hack is removed.

RTimothyEdwards, 2022-09-27:
This was corrected a long time ago by a workaround in open_pdks:  `sky130/custom/scripts/fix_gpiov2_gds.py`.  This ticket can be closed.

closed 2022-09-27 by RTimothyEdwards