efabless/caravel #44: iverilog syntax error on Caravel

emre-goncu, 2022-03-10:
Since my RTL code have some "typdef struct" statements, I've added the parameter "-g2005-sv" for iverilog to the related lines in the Makefile given in the path "caravel_user_project/mgmt_core_wrapper/verilog/dv/make/env.makefile". But after those changes, I get the below errors, while in the verification phase:

_caravel/verilog/rtl/caravel.v:614: error: Invalid module instantiation_

marwaneltoukhy, 2022-03-11:
@emre-goncu you added an iverlog parameter in the environment makefile? the ``env.makefile`` is nothing but a makefile that exports some global variables, that infact can be overwritten by exporting them from outside the makefile. I don't really understand your change.

emre-goncu, 2022-03-14:
Sorry about that! I wanted to mention "sim.makefile" not "env.makefile".

kauser-rl, 2022-03-24:
It might be related to [this](https://github.com/efabless/caravel/issues/34).

closed 2022-10-20 by RTimothyEdwards