efabless/caravel #47: user_id_programming layout is incorrect

d-m-bailey, 2022-03-21:
The `user_id_programming` block should have connections that reflect the user_id textblock, but instead they are all zero.

`caravel/scripts/set_user_id.py` creates a gds file with the correct connections, but the script that merges caravel blocks does not reference it. The following is from `caravel/Makefile`.
```
__ship:
        @echo "###############################################"
        @echo "Generating Caravel GDS (sources are in the 'gds' directory)"
        @sleep 1
#### Runs from the CARAVEL_ROOT mag directory
        @echo "\
                random seed `$(CARAVEL_ROOT)/scripts/set_user_id.py -report`; \
                drc off; \
                crashbackups stop; \
                addpath hexdigits; \
                addpath $(CARAVEL_ROOT)/mag; \
                addpath $(UPRJ_ROOT)/mag; \
                load user_project_wrapper; \
                property LEFview true; \
                property GDS_FILE $(UPRJ_ROOT)/gds/user_project_wrapper.gds; \
                property GDS_START 0; \
                load mgmt_core_wrapper; \
                property LEFview true; \
                property GDS_FILE ../mgmt_core_wrapper/gds/mgmt_core_wrapper.gds; \
                property GDS_START 0; \
                load $(UPRJ_ROOT)/mag/user_id_programming; \
                load $(UPRJ_ROOT)/mag/user_id_textblock; \
                load ../maglef/simple_por; \
                load $(UPRJ_ROOT)/mag/caravel -dereference; \
                select top cell; \
                expand; \
                cif *hier write disable; \
                cif *array write disable; \
                gds write $(UPRJ_ROOT)/gds/caravel.gds; \
                quit -noprompt;" > $(UPRJ_ROOT)/mag/mag2gds_caravel.tcl
```
The unmodified `mag` view is loaded. Maybe the following change (to both `__ship` and `__truck`) would be sufficient.
```
                load $(UPRJ_ROOT)/mag/user_id_programming; \
                property LEFview true; \
                property GDS_FILE $(UPRJ_ROOT)/gds/user_id_programming.gds; \
                property GDS_START 0; \
                load $(UPRJ_ROOT)/mag/user_id_textblock; \
```

RTimothyEdwards, 2022-03-21:
@d-m-bailey :  But the `user_id_programming.mag` file being loaded has a property `GDS_FILE` that (I think correctly) points to `../gds/user_id_programming.gds`.  So the correct GDS should be referenced.

RTimothyEdwards, 2022-03-21:
@d-m-bailey :  It is true that something went wrong with ChipIgnite-one and all the projects were showing up as zero when queried from the SPI.  But my spot-check of an MPW-four project showed that the project number was correctly encoded in the top-level GDS.

RTimothyEdwards, 2022-03-21:
@d-m-bailey :  It might be altogether better to have the script modify the .mag file instead of the .gds file, but that's a separate question.

d-m-bailey, 2022-03-21:
@RTimothyEdwards Thanks for taking the time to look it to this. I checked mpw4 slot-001, slot-002, and slot-003. The displayed project number is correct, but the actual contacts in the `user_id_programming` block all appear to be the same, i.e. zero.
Like you said, the `mag` file for `user_id_programming` does contain a reference for `../gds/user_id_programming.gds`, but since the script is executing in `$CARAVEL_ROOT/mag` it ends up pointing to an unmodified `$CARAVEL_ROOT/gds/user_id_programming.gds`. The modified gds is at `$UPRJ_ROOT/mag/user_id_programming.gds`. At least, that's how it appears to me.
```
__ship:
        @echo "###############################################"
        @echo "Generating Caravel GDS (sources are in the 'gds' directory)"
        @sleep 1
#### Runs from the CARAVEL_ROOT mag directory
        @echo "\
                random seed `$(CARAVEL_ROOT)/scripts/set_user_id.py -report`; \
                drc off; \
                crashbackups stop; \
                addpath hexdigits; \
                addpath $(CARAVEL_ROOT)/mag; \
                addpath $(UPRJ_ROOT)/mag; \
                load user_project_wrapper; \
                property LEFview true; \
                property GDS_FILE $(UPRJ_ROOT)/gds/user_project_wrapper.gds; \
                property GDS_START 0; \
                load mgmt_core_wrapper; \
                property LEFview true; \
                property GDS_FILE ../mgmt_core_wrapper/gds/mgmt_core_wrapper.gds; \
                property GDS_START 0; \
                load $(UPRJ_ROOT)/mag/user_id_programming; \
                load $(UPRJ_ROOT)/mag/user_id_textblock; \
                load ../maglef/simple_por; \
                load $(CARAVEL_ROOT)/mag/caravel -dereference; \
                select top cell; \
                expand; \
                cif *hier write disable; \
                cif *array write disable; \
                gds write $(UPRJ_ROOT)/gds/caravel.gds; \
                quit -noprompt;" > $(UPRJ_ROOT)/mag/mag2gds_caravel.tcl
### Runs from CARAVEL_ROOT
        @mkdir -p ./signoff/build
        @cd $(CARAVEL_ROOT)/mag && PDKPATH=${PDK_ROOT}/sky130A MAGTYPE=mag magic -noc -dnull -rcfile ./.magicrc $(UPRJ_ROOT)/mag/mag2gds_caravel.tcl 2>&1 | tee $(UPRJ_ROOT)/signoff/build/make_ship.out
```

RTimothyEdwards, 2022-03-22:
@d-m-bailey :  Ah, the error is that 
```
load $(UPRJ_ROOT)/mag/user_id_programming; \
load $(UPRJ_ROOT)/mag/user_id_textblock; \
```
should be

```
load $(CARAVEL_ROOT)/mag/user_id_programming; \
load $(CARAVEL_ROOT)/mag/user_id_textblock; \
```

d-m-bailey, 2022-03-22:
@RTimothyEdwards The `set_user_id` target in the `caravel/Makefile` copies the `$(CARAVEL_ROOT)` files before modifying them.
```
.PHONY: set_user_id
set_user_id: check-env check-uid uncompress uncompress-caravel
ifeq ($(FOREGROUND),1)
        $(MAKE) -f $(CARAVEL_ROOT)/Makefile __set_user_id
        @echo "Set user ID completed." 2>&1 | tee -a ./signoff/build/set_user_id.out
else
        $(MAKE) -f $(CARAVEL_ROOT)/Makefile __set_user_id >/dev/null 2>&1 &
        tail -f signoff/build/set_user_id.out
        @echo "Set user ID completed." 2>&1 | tee -a ./signoff/build/set_user_id.out
endif

__set_user_id:
        mkdir -p ./signoff/build
        # Update info.yaml
        # sed -r "s/^(\s*project_id\s*:\s*).*/\1${USER_ID}/" -i info.yaml
        cp $(CARAVEL_ROOT)/gds/user_id_programming.gds ./gds/user_id_programming.gds
        cp $(CARAVEL_ROOT)/mag/user_id_programming.mag ./mag/user_id_programming.mag
        cp $(CARAVEL_ROOT)/mag/user_id_textblock.mag ./mag/user_id_textblock.mag
        cp $(CARAVEL_ROOT)/verilog/rtl/caravel.v ./verilog/rtl/caravel.v
        python3 $(CARAVEL_ROOT)/scripts/set_user_id.py $(USER_ID) $(shell pwd) 2>&1 | tee ./signoff/build/set_user_id.out
```
The modified files are in `./gds` and `./mag` which is equivalent to `$(UPRJ_ROOT)/gds` and `$(UPRJ_ROOT)/mag`.

I imagine this was to keep things reentrant. I don't believe we want to modify the originals in `$(CARAVEL_ROOT)`, but what do you think.

Also, would be nice if the `verilog/gl/user_id_programming.v` file was also modified. Currently, only `verilog/rtl/user_id_programming.v` is.

RTimothyEdwards, 2022-03-22:
@d-m-bailey :  I have just created a pull requiest to both fix the problem with the GDS and to modify the gate level verilog corresponding to the ID value change.

d-m-bailey, 2022-04-21:
Working now.

closed 2022-04-21 by d-m-bailey