efabless/caravel #54: "make ship" and "make truck" are missing dependances

derekcom17, 2022-04-03:
These targets will fail if several other tasks have not been completed first. No helpful error message is produced.

One way to resolve this is to add the dependent targets `check-uid check-mcw set_user_id gpio_defaults` to `ship` and `truck`.

Also, the path `../mgmt_core_wrapper/gds/mgmt_core_wrapper.gds` which appears in both of these targets could be replaced with `$(MCW_ROOT)/gds/mgmt_core_wrapper.gds`. That hardcoded path requires that MCW_ROOT be a subdirectory of CARAVEL_ROOT.

Pramodgouda1, 2022-07-28:
make gpio_defaults
export USED_ID=<your favorite 8 digit hex number>
make set_user_id
make ship

This should work. @derekcom17

derekcom17, 2022-07-29:
Yes I believe that should work.
If that is the intended script for `make ship`, then could you consider putting that script in the readme or somewhere obvious to new users?