efabless/caravel #72: gpio_default_block not working for pins 0-4

d-m-bailey, 2022-04-18:
From `caravel_user_project`
```
make gpio_defaults
```
gives
```
Traceback (most recent call last):
  File "/home/kanobailey/mpw-5/mpw-5d/caravel_user_project/caravel/scripts/gen_gpio_defaults.py", line 340, in <module>
    cellname = cellsused[int(gpioidx)]
ValueError: invalid literal for int() with base 10: '0\\[0\\]'
```
This is because the `mag` instances names for the first 5 `gpio_default_blocks` have changed.
```
use gpio_defaults_block  gpio_defaults_block_0\[0\]
use gpio_defaults_block  gpio_defaults_block_0\[1\]
use gpio_defaults_block  gpio_defaults_block_2\[0\]
use gpio_defaults_block  gpio_defaults_block_2\[1\]
use gpio_defaults_block  gpio_defaults_block_2\[2\]
```
Reverting these to the previous definitions
```
gpio_defaults_block_0
gpio_defaults_block_1
gpio_defaults_block_2
gpio_defaults_block_3
gpio_defaults_block_4
```
Allows the program to complete, but the blocks are not changed in `caravel.mag`.
This is because the program has been modified to skip the first 5 gpio_default_blocks.
```
    for i in range(5, 38):
        config_name = '`USER_CONFIG_GPIO_' + str(i) + '_INIT'
```
Reverting this to handle all 38 blocks corrects the problem.

Pull request will be submitted soon.

RTimothyEdwards, 2022-04-18:
This has to be re-checked in light of a pull request that I made to fix the issue five days ago but which was not merged.  I changed the script to recognize the array names and map them to indexes.  That should fix this issue so the pull request should not be needed.

d-m-bailey, 2022-04-18:
@RTimothyEdwards I'll back out the changes related to `gpio_defaults_block`.

PR #73 (2022-04-19): Changes that correct mgmt_core references and user_id_programming generation.
mgmt_core_wrapper: fixes #71 
- Use absolute path instead of relative path.

user_id_programming: fixes #70 
- Remove GDS references as GDS is no longer modified.
- Corrected string concatenation.
- Corrected mag data replacement.
- Corrected verilog data replacement.

<s>gpio_default_block: fixes #72~
- Rename instances for gpio_default_blocks 0-4 in caravel.mag and caravan.mag.
- Change replace range in gen_gpio_defaults.py to handle gpio_default_blocks 0-4.</s>
closed 2022-04-19 by jeffdi