efabless/caravel #85: Older versions of gpio_default_block_1803 and gpio_default_block_0403 appear to be used in make ship

d-m-bailey, 2022-04-23:
During `make ship` and `make truck`, `mag` data is read from `caravel_user_project/mag` and `caravel_user_project/caravel/mag`. Older versions of `gpio_default_block_1803` and `gpio_default_block_0403` exist in `caravel/mag` and appear to override the newer versions in `caravel_user_project/mag`.

To reproduce, in `caravel_user_project`
```
make ship
```
The output gds should match the `caravel_user_project/mag` data, but I think it's `caravel_user_project/caravel/mag` data. 

LVS doesn't match for these 2 cells and the mag data is different, so I'm assuming this is the reason. 

Will submit a pull request to remove the mag data from `caravel/mag`.

There are many of the following messages generated during `make ship`, which indicates that the location of the data is very relevant when there are versions in both directories.
```
Warning:  Parent cell lists instance of "sky130_fd_sc_hd__and2b_2" at bad file path ../mag/sky130_fd_sc_hd__and2b_2.mag.
The cell exists in the search paths at $HOME/mpw-5/mpw-5d/pdks/sky130A/libs.ref/sky130_fd_sc_hd/mag/sky130_fd_sc_hd__and2b_2.mag.
The discovered version will be used.
Warning:  Parent cell lists instance of "gpio_defaults_block" at bad file path $HOME/mpw-5/mpw-5d/caravel_user_project/mag/gpio_defaults_block.mag.
The cell exists in the search paths at ../mag/gpio_defaults_block.mag.
```

d-m-bailey, 2022-04-23:
The problem is not with the gds, it seems investigating futher...

d-m-bailey, 2022-04-23:
The problem is that the verilog is not being updated correctly for the first 5 gpio_defaults_blocks
```
    gpio_defaults_block gpio_defaults_block_29 (                         |    gpio_defaults_block_1808 gpio_defaults_block_29 (                   
      .VGND(vssd_core),                                                  |      .VGND(vssd_core),
      .VPWR(vccd_core),                                                  |      .VPWR(vccd_core),                                                 
      .gpio_defaults({ \gpio_defaults[389] , \gpio_defaults[388] , \gpio_|      .gpio_defaults({ \gpio_defaults[389] , \gpio_defaults[388] , \gpio
    );                                                                   |    );
    gpio_defaults_block \gpio_defaults_block_2[0]  (                     |    gpio_defaults_block \gpio_defaults_block_2[0]  (                    
      .VGND(vssd_core),                                                  |      .VGND(vssd_core),
 ```
You can see the instances that we're updated and those that weren't.

Patched `scripts/gen_gpio_defaults.py`. Testing now.

RTimothyEdwards, 2022-04-23:
Thanks for investigating.  Let me know when you have a pull request, or if you need any help.

PR #87 (2022-04-24): Fix verilog gpio_defaults_block replacement for gpio 0-4
Fixes #85
closed 2022-04-24 by jeffdi