efabless/caravel #99: chip_io.def contains 0.5nm real coordinate despite UNITS .. MICRONS 1000

stefanottili, 2022-05-28:
Even though it seems that the si2 based def parsers reads this def file, it leads to ambiguity.

a) what tool writes coordinates as real numbers ?
b) why write 0.5um coordinates when the units are set to 1000dbu (1=1nm) ?

   6    UNITS DISTANCE MICRONS 1000 ;
7667       NEW met3  ( 3354120 2322992.5 ) ( 3384240 * )

RTimothyEdwards, 2022-05-29:
(a) Probably magic.
(b) Probably obscure reasons.

Sometime recently, someone did a search on how to read an arbitrary layout into DEF using open source tools, and the only useful result showed how to do it in magic.  So apparently magic is the only open source tool that can do this.  I wrote the LEF/DEF parser and generator decades ago but never had any particular use for the DEF generator, so it has remained in a development state for many years.  It did manage to do the job of converting the partly hand-placed, hand-routed padframe design back into a DEF file, though.  Some quirks are to be expected.

closed 2022-05-30 by stefanottili

RTimothyEdwards, 2022-05-30:
@stefanottili :  I'd appreciate it if you could open this as an issue in github.com/RTimothyEdwards/magic, for the issue of generating an off-grid point.  I put in code to write out a fractional value to be able to catch any instance like this of magic thinking a point is not on the grid.  Everything in the design _must_ be on-grid by definition in magic, so I don't know what it thought it was doing when it generated that value.

RTimothyEdwards, 2022-05-30:
@stefanottili :  And yes, the problem that I was trying to solve when I generated the DEF was that I had modified parts of the routing manually and wanted to read the DEF file back into OpenROAD with the manual modifications, and at least for the purpose needed, the DEF output from magic was good enough.