efabless/caravel #105: caravan.spice file used for pre-check is missing top_level connections

d-m-bailey, 2022-07-15:
The pre-check uses `caravel/spi/lvs/caravan.spice` to check the connections to `user_analog_project_wrapper`.

It is expecting `vccd1_core` to be connected to `user_analog_project_wrapper/vccd1` but the netlist has `mprj/vccd1` connected instead.

All instances in the top level caravan are not connected to any other instances. Maybe the top level wiring was left out or maybe a bad version of magic was used.
Example
```
Xgpio_defaults_block_7 gpio_defaults_block_7/VGND gpio_defaults_block_7/VPWR gpio_defaults_block_7/gpio_defaults[0]
+ gpio_defaults_block_7/gpio_defaults[10] gpio_defaults_block_7/gpio_defaults[11]
```

jainsoumil2, 2022-08-29:
Will this also error on the precheck on efabless website?

carllb, 2022-08-29:
It seems that the error is present in both the local mpw-precheck and the efabless website for analog projects.

jeffdi, 2022-09-27:
Have Risto look at this to see if it is already fixed.