efabless/caravel #107: [Question] use case of digital pll

sajjadahmed677, 2022-08-04:
Hi all,
I have seen a ```digital PLL``` block in caravel repo so i have question regarding this. is this PLL actually being used in caravel or it is just a behavioral model of actual ```analog PLL``` for verification purpose. meaning does caravel have actually an ```all digital PLL```.

RTimothyEdwards, 2022-09-27:
Other than the fact that it is a frequency-locked loop and not a phase-locked loop, yes, caravel has an actual DLL (digital locked loop).  The measured range of the DLL on Caravel MPW-two is minimum 38.8 MHz, maximum 95.2 MHz.  The output can be divided down by 1 to 7 times (so the minimum clock rate derived from the DLL is 5.54 MHz).  On the feedback side, the DLL output can be divided by 2 to 32 times, and so can lock to an input clock of 1.66 MHz to 47.6 MHz.

DavidRLindley, 2024-12-10:
Closing because I think the original question was answered.