efabless/caravel #108: mismatch between gds and mag views

kareefardi, 2022-08-07:
currently I am aware of:
- `mgmt_protect`

I will update this list whenever I find more mismatches.

RTimothyEdwards, 2022-09-02:
Why is there even a GDS view of this block, since it's not used?  The gds/ directory is full of GDS files that are not used and should be deleted.  The only GDS files that I can see that are actually being used are `advSeal_5um_gen.gds` and `simple_por.gds`.

Both of those should be rewritten with `.gz` at the end in the corresponding `.mag` files, and we should stop compressing/decompressing GDS files outside of the layout tools.

RTimothyEdwards, 2022-09-27:
Reassigning to Kareem.  This requires only removing all of the unused GDS files (which is everything except those mentioned above) and confirm that this does not break the design and tapeout flow.  The scripts should be rewritten to force magic to always use `gds compress 9` before writing GDS, and then to remove all compress/uncompress scripts from the repository (except possibly for generating the final tapeout database, depending on whether a foundry can or cannot read compressed GDS, or will be using only the Oasis database) (Global Foundries, for example, currently will take only uncompressed GDS).

azwefabless, 2022-10-14:
Tapeout script specifies a single source of truth for each block and we will archive it as we tapeout with a tag. No need to work on this ticket as it represents an old status of the repo based on prior practices.