efabless/caravel #111: gpio_control_block.def is out of date

antonblanchard, 2022-08-30:
Commit b2089fe9eb2d updated the mag and gate level views, but the def was not. This breaks hierarchical STA (eg `make caravel_timing_typ`).

Note: The spef view was also not updated, but at least it can be regenerated simply via `rcx-gpio_control_block`

RTimothyEdwards, 2022-09-02:
There are multiple issues that need to be resolved with `gpio_control_block`;  for one, the timing issue between control blocks has been fixed but is currently in branch `fix_serial_loader_data_timing`.  I have not issued a pull request because this block needs to be re-synthesized and verified, and all views (including DEF, LEF, .mag, gate-level verilog) updated.  Then it can be merged into main.

@shalan :  Please make sure that this block gets rehardened.  This is a critical fix.

shalan, 2022-09-07:
@antonblanchard I am afraid to say that hierarchical STA does not work anymore as intended due to limitations in the current OpenSTA release. Please don't rely on it as the top-level interconnect parasitics are not used/seen by OpenSTA. This has been confirmed by the OpenROAD team.

RTimothyEdwards, 2022-10-05:
This will be replaced by a new DEF, since the block needs to be rehardened.

closed 2022-10-22 by marwaneltoukhy