efabless/caravel #112: Package information for PCB design

suzizecat, 2022-09-05:
Hello,

Unless I missed it somewhere, the package information about the WLCSP used for Caravel seems to be missing.
I found https://caravel-harness.readthedocs.io/en/latest/pinout.html but this doesn't provide the bump size nor the recommended pad size and soldermask margin.

One of those two information is required in order to design PCBs using Caravel.
It is particularly important as the dimension provided are already really tight with regard to "standard" cheap PCB manufacturer capabilities. (JLCPCB, PCBWay, Aisler and probably others).

Could it be possible to provide the solder bump size of the WLCSP package ? (I would love to have the recommended pad size, but I would understand that you don't want to take responsibility with regard to PCB assembly.)

Best regards,
Julien FAUCHER

RTimothyEdwards, 2022-09-27:
Please see:  https://github.com/efabless/caravel/blob/main/docs/source/_static/bond_plan.svg
The bump pitch is 0.5mm.  Note that the diagram is out of date (I have just corrected it and pushed the new version), and the bump diameter should be shown as 350um.

Also see https://github.com/efabless/caravel/blob/main/docs/source/_static/pcb_example_route_pattern.svg
for the recommended PCB pattern to reach all pins using a standard cheap PCB process (it is, admittedly, still rather tight, and you will need to shave the size of the pads down to less than the optimal dimension, but you should still be able to get a reliable attachment).

suzizecat, 2022-09-28:
Hi @RTimothyEdwards , 

Thanks for your reply. On a personal note, I think it might be interesting to have this information in the table 3 of https://caravel-harness.readthedocs.io/en/latest/pinout.html (at the end of the page) to have all information in one table.

Anyway, the few documentation I found about WLCSP recommendations [Analog Device AN-617](https://www.analog.com/media/en/technical-documentation/application-notes/AN-617.pdf) or [This website](https://www.eeweb.com/wafer-level-chip-size-package-wlcsp-guidelines/) mention a bump diameter of 320um for a bump pitch of 0.5mm .

While I'm not sure that 20um would have a very significant impact, it is always nice to be precise. Can you confirm that the bump diameter is 350um and not 320um ?

> Also see https://github.com/efabless/caravel/blob/main/docs/source/_static/pcb_example_route_pattern.svg
> for the recommended PCB pattern to reach all pins using a standard cheap PCB process (it is, [...]

Just to be sure I am clear on my explanations, I was not asking for the escape route (that you provide and is a really nice thing to have ) but the pad size (see the previously linked  [AN-617](https://www.analog.com/media/en/technical-documentation/application-notes/AN-617.pdf), page 5 for an example. 

I was wondering if the company that performs the packaging (the packager ?) had such reference document about "their" WLCSP available.

RTimothyEdwards, 2022-09-28:
@suzizecat :  Bump diameter depends on the capabilities of the vendor doing the bump bonding.  I wanted a bump diameter of 250um but 350um was the minimum diameter offered by Micross.

Micross does have some documentation, but they are a small vendor and do a lot of custom work, so they do not have any "fixed" process but will build to the specification of the customer.  There are some minimum/maximum limits such as redistribution layer wire width and spacing, and minimum bump diameter, but that's all.

The layout of the redistribution layer is part of the caravel chip design, but you have to go back to the original repository, which is now called `caravel_mpw-one`, to find the files "caravel_bump_bond.mag" and "caravel_bump_bond.gds.gz".  From the GDS file you can get the pad size and the dimensions of the redistribution metal layer.

suzizecat, 2022-09-28:
@RTimothyEdwards 

> Bump diameter depends on the capabilities of the vendor doing the bump bonding. I wanted a bump diameter of 250um but 350um was the minimum diameter offered by Micross.

Therefore, it is definitely 350um, thanks !

> The layout of the redistribution layer is part of the caravel chip design, but you have to go back to the original repository, which is now called `caravel_mpw-one`, to find the files [...]

I think we might have a misunderstanding here, you seems to be talking about pads on the die to the bounding (or to the bumps) while, in this discussion, I am thinking about PCB Footprint pads. My original question being about the dimensions to take into account when designing the PCB board that will receive the caravel "component".

RTimothyEdwards, 2022-09-28:
@suzizecat :  This is WLCSP, with the solder bumps attached directly to the front side of the chip die, so the position of the pads on the chip redistribution metal layer exactly matches the position of pads on the PCB (only flipped around one axis).