efabless/caravel #114: housekeeping_spi idata needs a careful attention (Fast to Slow CDC)

shalan, 2022-09-09:
The `housekeeping_spi` `idata` port is in synch with `wb_clk_i` clock; yet it is written to a register with the `CSB` clock. As the `CSB` clock is slower than `wb_clk_i`, a synchronizer is not needed. However, the design has to be carefully reviewed to ensure that `idata` will be captured inside the `housekeeping_spi` without any issue.

RTimothyEdwards, 2022-09-27:
The housekeeping_spi `idata` port (`odata` in housekeeping.v) is not synchronized to anything;  it is a combinatorial assignment that depends only on the address word `caddr`.

azwefabless, 2022-10-14:
Addressed this issue by forcing the clocks to be mutex for htis iteration of caravel by holding CPU in reset while using the PSI interface.