efabless/caravel #115: housekeeping clock "csclk" isn't synthesised as a clock

M0stafaRady, 2022-09-11:
I can see from sdf simulations that the leaves of csclk are having large delays relative to mgmt_gpio_in[4] 
`

     assign csclk = (wbbd_busy) ? wbbd_sck : ((spi_is_active) ? mgmt_gpio_in[4] : 1'b0);

`
when I checked the sdc file for the housekeeping I found a generated clock named csclk_slow but I can't find csclk however gatelevel netlist has csclk net and nothing name csclk_slow.

My guess is that some constrains might need to be added/updated. 

> delays seen in simulation 
> clknet_2_3_0_mgmt_gpio_in[4] to csclk = 443 ps 
> csclk to clknet_leaf_35_csclk =  1139 ps

RTimothyEdwards, 2022-09-11:
I would not necessarily call 1 1/2 nanoseconds a large delay, especially relative to the core clock minimum period of 25 ns.  What matters is the clock to data timing at the flops, in particular the timing of csclk to cdata.

The generated clock names are unique to the SDC file.  The `csclk` does not appear in the SDC file presumably because the `csclk` is an internal wire and it is optimized out of the gate-level verilog.  But the references in the SDC file to instance names is very fragile and must be updated any time the block is re-hardened.  Those type of references should be avoided except as a last resort.  For example, `_9239_/Q` is a named net (`serial_clock_pre`), so why isn't the net name used there?  Likewise, `_9640_/Q` is `wbbd_sck`.

kareefardi, 2022-09-11:
`csclk` net has a clock tree in the current implementation. it is specified in `CLOCK_NET` which ultimately what was used by cts at the time see:  https://github.com/efabless/caravel/blob/9949306c42ded3cad03e1f4566d7cd1f8215b0bb/openlane/housekeeping/config.tcl#L29 also the fact that flip-flops driven by `csclk` have clock nets similar to `clknet_leaf_35_csclk`. i don't think that the generated clock name in the sdc file influence the clock net. regarding the implementation itself of the clock tree and the part before `csclk` itself is another story

M0stafaRady, 2022-09-12:
@RTimothyEdwards - I can see timing violation because of this 1.5 ns delay in the data driven by clknet_2_3_0_mgmt_gpio_in[4] flip-flop and suppose to be captured by clknet_leaf_35_csclk  flip-flop. Data changes 744 ps before clknet_leaf_35_csclk positive edge. 

> clknet_2_3_0_mgmt_gpio_in[4] positive edge to cdata change = 25830 ps 
> clknet_leaf_35_csclk positive edge to cdata change = 24256 ps 

@kareefardi - that makes sense but do you know why there are 5 buffers between `csclk ` and `clknet_leaf_35_csclk `

kareefardi, 2022-09-13:
No I don't know

passant5, 2022-10-14:
-name <clock_name> in the SDC only defines the name of the clock group and doesn't affect the clock synthesis

closed 2022-10-14 by passant5

azwefabless, 2022-11-01:
Fixed in SDC but needs documentation to close.