efabless/caravel #149: Add some predefined Spare cells to the RTL definition of the GPIO control logic so that it gets instantiated per IO.

azwefabless, 2022-10-05:
Something like 2 Flops, 8 buffers, 8 inverters, 8 2-input nand gates, some constants.

RTimothyEdwards, 2022-10-06:
The simplest implementation is to change the number of instances of `spare_logic_block` in caravel.v and caravan.v from 4 to 42, and place one spare logic block for each gpio_control_block in the padframe.  Probably it would help to resynthesize the spare_logic_block to make it a bit smaller.  It does not need the decap (which is taking up half or more of the cell area), and there should be virtually no routing congestion.  The spare logic block should be routed without use of local interconnect, because that defeats the purpose of the spare logic block to be used for metal-mask-only back-end-of-line fixes.

RTimothyEdwards, 2022-10-07:
See pull request #153.

closed 2022-10-20 by RTimothyEdwards