efabless/caravel #159: caravel does not pass LVS

d-m-bailey, 2022-10-08:
This is a list of the problems encountered (and workarounds) to run full device level LVS on an empty instance of caravel using `run_full_lvs` of the `caravel` branch of https://github.com/d-m-bailey/extra_be_checks.git.

1. Add `SUBCUT` to `gpio_control_block` layout around `gpio_logic_high`.
2. Modify the `chip_io` layout cells according to https://github.com/efabless/caravel/pull/29
3. The above layout modifications require patched versions of `sky130_ef_io.spice` and `sky130_fd_io.spice` which are included in `extra_be_checks/spice`.
4. Remove property comparison for `res_generic_po`, `res_generic_li`, and `res_generic_m*`. `chip_io` circuits use the same subcells in schematic with sizes that don't correspond to multiple different layouts. Fix is in `extra_be_checks` setup file.
5. Ignore `__fill_*`, `__tapvpwrvgnd_`, and `__fakediode_` when comparing. These cells are not included in some of the `caravel` blocks. Fix is in `extra_be_checks` setup file.
6. Create `user_project_wrapper.v` gl verilog from `__user_project_wrapper.v` rtl verilog. Fix is in `extra_be_checks/verilog`.
7. Patch `sky130_fd_sc_hd.spice` and `sky130_fd_sc_hvl.spice` with correct diode `pj` parameters. Patched files are in `extra_be_checks/spice`.
8. Change `vdda1_core`, `vdda2_core`, `vssa1_core`, `vssa2_core`, `vccd1_core`, `vccd2_core`, `vssd1_core`, `vssd2_core`,  `vssd_core`, and `vccd_core` from wires to pins. The layout has text and with cifinput(vendor), text are converted to pins.
NOTE: `vdda_core`, `vssa_core`, `vssio_core`, and `vddio_core` remain as wires. Will submit a pull request for an (untested) patch to the rtl.

To run:
```
git clone -b caravel https://github.com/d-m-bailey/extra_be_checks.git
export LVS_ROOT=$PWD/extra_be_checks
export LOG_ROOT=<the log root>
export SIGNOFF_ROOT=<the log root>
export WORK_ROOT=<the work root>
export CARAVEL_ROOT=<the caravel root>
export MCW_ROOT=<the mcw root>
$LVS_ROOT/run_be_checks caravel <gate level caravel verilog> <gds file>
```
Default referenced files are listed in `$WORK_ROOT/spice_files` and `$WORK_ROOT/verilog_files`.

I don't know how to submit pull requests for layout or spice files.

marwaneltoukhy, 2022-10-08:
Thank you @d-m-bailey , point 1 is done and currently on `caravel_redesign` branch

RTimothyEdwards, 2022-10-20:
Caravel passed LVS a few days ago, so this issue will be closed.
However, for the record, note that all required changes to library cells are PDK issues, and tickets are being generated in the open_pdks repository to handle them.  Meanwhile, we are using a method of getting the design LVS clean on the top level with abstracted I/O cells, and passing that design off to Mitch to apply PDK changes and verify that the design passes LVS down to the device level.

This issue will be renamed to reflect that LVS is still not passing at the device level using the current PDK, and will remain open until such time as we can run LVS at the device level with no special modifications.  It is still a high priority item but currently does not block tapeout.