efabless/caravel #164: simple_por lef view is not matching the gds

kareefardi, 2022-10-10:


kareefardi, 2022-10-10:
I didn't not check the layout. What I know is that the gds and lef belong to different commits (which isn't necessarily an issue). However using the current lef leads to drcs during the toplevel routing which get removed upon regenerating the lef (again doesn't mean a 100% that the views aren't matching but likely they are).

RTimothyEdwards, 2022-10-10:
I see that the LEF file is a detailed view of the layout, which it shouldn't be;  the abstract view in `maglef/` is what I would expect the LEF to look at;  obviously the `maglef/` view was generated from a different LEF file.  I should have that LEF file somewhere locally or else I can regenerate it.  I will make a pull request for it.

kareefardi, 2022-10-10:
rename the issue to reflect the state

RTimothyEdwards, 2022-10-11:
@kareefardi :  PR #170 should fix this problem.

closed 2022-10-20 by RTimothyEdwards