efabless/caravel #191: Documentation issues and enhancements identified from config.tcl.

azwefabless, 2022-10-12:
CLK_WIRE_RC_LAYER still says Met5

CLK_BUFRFER, CLK_BUFFER_INPUT, CLK_BUFFER_OUTPUT are not documented in env doc on caravel openlane
This needs to become an openlane ticket as there are many issues like this.

Some blocks seem to be implemented in Openlane using ant strategy 3 and some using ant strategy 4 with no clear documentation or reasoning.

TS_ROOT_BUFFER not documented

CTS SQR CAP, CTS_SQR_RES not documented and values do not seem consistent with Met 5 in item1 above.

CTS_TECH_DIT is N/A and item is undocumented.

Range for CTS TOLERANCE is not defined in documentation.

DATA_WIRE_RC_LAYER is not documented. Setting of M2 seems reasonable if it is used for Groute estimation.

DEF_UNITS_PER_MICRON 1000 (surprised that this is not 200). Undocumented. Creates he opportunity for offgrid placement errors.

DESIGN_CONFIG seems to reference itself.

DPL_CELL_PADDING is not documented

DRC_EXCLUDE_CELL_LISTt is pointed to a design specific cell.
It seems to be being used to solve problems that are unrelated to routing DRC.Probably need a different solution t this problem.

DRC_EXCLUDE_CELL_LIST_OPT is pointed to the pdk default cell list.

ECO_ENABLE, ECO_FINISH and ECO_ITER and ECO_SKIP_PIN are undocumented.

FAKEDIODE_CELL is undocumented

FILL_CELL is undocumented

HLENGTH seems like a crude routing solution which produces inefficient layouts. Future enhancement
VLENGTH Is same issue

FP_IO_HTHICKNESS_MULT seems very odd and is not a routed wire pitch. Future enhancement

FP_IO_MIN_DISTANCE functionality needs improvement. Not connecting on routed pitches and consuming a lot of distance for amount of connectivity

IO_MODE default is set to 1 we need to understand the other modes or options. Future enhancement

FP_PDN_COORE_RING _* are undocumented.

ENABLE_GLOBAL_CONNECTIONS is undocumented.

FP_PDN_IRDROP needs clarifying documentation and a result.

PDN_LOWER_LAYER is not documented

PDN_RAILS_* not documented

PDN_UPPER_LAYER not documented

Stopped at row 132 of config.tcl for mgmt_soc

azwefabless, 2022-10-12:
This is the documentation related items that were originally part of ticket 188 and have been split out as they are not as urgent to address as the actionable items on the implementation script itself.

kareefardi, 2023-05-28:
- DATA_WIRE_RC_LAYER and CLOCK_WIRE_RC_LAYER are merged to WIRE_RC_LAYER which is set to met1. They should be back to two variables
- Antenna strategies deprecated and separated to more descriptive variables
- CLK_BUFFER* not used - should remove
- CTS_ROOT_BUFFER documented
- CTS SQR CAP and RES are not used - should remove
- CTS_TECH_DIR not used - should remove
- CTS_TOLERANCE not used - should remove
- DEF_UNIT_PER_MICRON not used - information fetched from techlef - should remove
- DPL_CELL_PADDING  documented
- DRC_EXCLUDE_CELL_LIST documented - confusing name should change
- _OPT variables - wip
- ECO* removed
- FAKEDIODE removed
- FILL_CELL documented
- IO_MODE - needs work
- FP_PDN_COORE_RING  documented
- ENABLE_GLOBAL_CONNECTIONS  documented
- FP_PDN_IRDROP removed - new variable wip
- PDN_(LOWER|UPPER|RAILS) documented

- Can you elaborate on HLENGTH, VLENGTH, MULT and MIN_DISTANCE

kareefardi, 2023-08-03:
Removed and was undocumented:
- CLK_BUFFER
- CLK_BUFFER_INPUT
- CLK_BUFFER_OUTPUT
- CTS_SQR_RES
- CTS_TECH_DIR
- ECO_ENABLE
- ECO_FINISH
- ECO_ITER
- ECO_SKIP_PIN
- FAKEDIODE_CELL

Removed - is documented or added documentation to with removed status:
- CTS_TOLERANCE

Documented:
- CTS_CLK_BUFFER_LIST
- DEF_UNITS_PER_MICRON - 1000 is default value in the techlef of the PDK. and is mostly unused:
- DPL_CELL_PADDING
- FILL_CELL
- ENABLE_GLOBAL_CONNECTIONS - Now FP_PDN_ENABLE_GLOBAL_CONNECTIONS
- PDN_LOWER_LAYER
- PDN_RAILS_*
- PDN_UPPER_LAYER

Internal Variable used by the flow:
- DESIGN_CONFIG - it references the config file of the design.

Actionable:
- DRC_EXCLUDE_CELL_LIST - bad name should rename to something more meaningful.
- CLOCK_WIRE_RC_LAYER and DATA_WIRE_RC_LAYER were merged into WIRE_RC_LAYER. They need to be split again.
- IO_MODE. Although it is documented, OpenROAD has changed and this needs to be revisited.
- FP_IO_HTHICKNESS_MULT - might look into making it pitch / track friendly.

Additional comments:
- FP_IO_MIN_DISTANCE - takes routing tracks into account.
- RUN_IRDROP_REPORT now is used instead of FP_PDN_IRDROP. It's an OpenROAD feature. Maybe needs a supporting document by OpenLane ?
- OpenLane PRs related to documentation:
    - https://github.com/The-OpenROAD-Project/OpenLane/pull/1917
    - https://github.com/The-OpenROAD-Project/OpenLane/pull/1887
    - https://github.com/The-OpenROAD-Project/OpenLane/pull/1891
    - https://github.com/The-OpenROAD-Project/OpenLane/pull/1764
    - https://github.com/RTimothyEdwards/open_pdks/pull/390
    - https://github.com/The-OpenROAD-Project/OpenLane/pull/1928

closed 2023-08-23 by jeffdi