efabless/caravel #203: `standalone_pvr.report` under `signoff/<design>/`

marwaneltoukhy, 2022-10-13:
Add a standalone PVR report for each block, this includes:

1. LVS
2. DRC
3. CVC
4. Antenna

This can be done using the automation script.

passant5, 2022-10-22:
- LVS, DRC, and STA of all re-implemented blocks and top-level caravel are added to `./signoff/<design_name>/signoff.rpt` using Caravel [signoff automation script](https://github.com/efabless/caravel/blob/caravel_redesign/scripts/signoff_automation.py)
- CVC is performed using Mitch's custom script and it runs on the caravel top-level
- Magic antenna checks never finished (killed after 48 hours); Calibre antenna checks require runset development.

jeffdi, 2022-11-01:
fixed in b6288a75d70831da3602a3fd94d8826656ef2e91

closed 2022-11-01 by jeffdi