efabless/caravel #208: DRC deck is a subset of the deck intended by the technology developers (Needs review)

azwefabless, 2022-10-13:
*****33) We are not running the full S8 development DRC deck and need to review what rules are being missed. 
(At minimum we need a review before final tapeout.)

jeffdi, 2022-10-13:
to be done before tapeout

azwefabless, 2022-10-20:
Reviewed the Implemented rule vs overall rules set in earliest deck documentation I had access to through SKY. While DRC deck is incomplete and should be reviewed before next tapeout the rules missing are mostly around specific components I know are not being used in caravel. We need to check this short component list for user project. (Specifically which MIM and Cap components are included in the metal stack and do we include all of the relevant DRC rules for those components allowed. To do before User Project Tapeouts. Review with Jeff