efabless/caravel #229: update caravel constraints to limit false path for mprj_io

jeffdi, 2022-10-14:
The following false path statement needs to be modified to limit it to the path from the pad to housekeeping (and not the user project wrapper).

set_false_path -from [get_ports mprj_io[*]]

example 

set_false_path -from [get_ports mprj_io[*]] -to ...

passant5, 2022-10-16:
updated the constraints sdc to add false paths from the pads to housekeeping pins except for `housekeeping/mgmt_gpio_in[4]` (SCK) and `housekeeping/mgmt_gpio_in[2]` (SDI)

example:
 
```
set_false_path -from [get_ports mprj_io[0]] -through [get_pins housekeeping/mgmt_gpio_in[0]]
set_false_path -from [get_ports mprj_io[*]] -through [get_pins housekeeping/mgmt_gpio_out[*]]
set_false_path -from [get_ports mprj_io[*]] -through [get_pins housekeeping/mgmt_gpio_oeb[*]]
```

closed 2022-10-16 by passant5

azwefabless, 2022-11-01:
This was edited again later in the design to add/eliminate specific MPRJ paths to the false set.

passant5, 2022-11-01:
commit https://github.com/efabless/caravel/pull/250/commits/dfdfea37780edbc8b6de0575caaf43387ac23902