efabless/caravel #231: mgmt_protect_hv power pins

mo-hosni, 2022-10-14:
- Labels and power pins swapped in gds. The actual pin rectangles' layer is label, while the text's layer is pin.
- Power pins' widths may be increased from 300nm to 500nm so that vias can be inserted automatically from the top PDN to the power pins.

RTimothyEdwards, 2022-10-14:
@mo-hosni :  The layout will remain the same for this subcircuit?  In that case I can just read the GDS, correct the pins, and rewrite it.  Aren't all of the subcells of `mgmt_protect` incorrect in the same way?

RTimothyEdwards, 2022-10-14:
Okay, I see that (at least as of now) only the cell you mention has this issue.  The LEF view looks correct, so I am only updating the GDS and .mag views.

RTimothyEdwards, 2022-10-14:
@mo-hosni :  Files have been updated an pushed.  Needs review and merge.  No mask layout is changed.  I just converted the labels back to pins, removed the redundant single-point labels, and regenerated GDS along with saving the new .mag file.

mo-hosni, 2022-10-14:
@RTimothyEdwards 
Great, that fixed the first issue.
Can you change the horizontal power pins width to be 500nm instead of 300nm?

RTimothyEdwards, 2022-10-14:
@mo-hosni :  Done and pushed.

RTimothyEdwards, 2022-10-14:
@mo-hosni :  New LEF file created for the re-sized horizontal metal3 lines, and pushed.

closed 2022-10-15 by mo-hosni