efabless/caravel #302: 1.8V buffer on 3.3V signal.

d-m-bailey, 2022-10-19:
`/por/porb_h`, a 3.3V signal is passed through a 1.8V buffer `/Xsoc(mgmt_core_wrapper)/Xinput207(PX_sky130_fd_sc_hd__clkbuf_16)` before being sent as `/soc/porb_h_out` to the `gpio`/`xres` blocks, which appear to be expecting a 3.3V signal.

d-m-bailey, 2022-10-19:
With `caravel_redesign` branch: commit `bbb6bf775c6c2c31376d5992906f929953aaef0b` using `scripts/signoff_automation.py` to create `caravel.gds`, it appears that 3.3V `porb_h` in to 1.8V `soc` buffer has been fixed. 

net trace of `simple_por` output shows connections directly to gpio cells and no connections to `soc`.

passant5, 2022-10-20:
Fixed here: https://github.com/efabless/caravel/issues/273

closed 2022-10-20 by RTimothyEdwards

azwefabless, 2022-11-01:
This was completed as a direct commit to caravel_redesign. Not best practice but is done.