efabless/caravel #369: `gpio_signal_buffering` parastic extraction fail

kareefardi, 2022-11-07:
script used to extract:
```
#!/bin/env bash

set -uex

if [[ $PDK == "sky130A" ]] ; then
    fd_cells_lef=${PDK_ROOT}/${PDK}/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef
    ef_cells_lef=${PDK_ROOT}/${PDK}/libs.ref/sky130_fd_sc_hd/lef/sky130_ef_sc_hd.lef
    tech_lef=${PDK_ROOT}/${PDK}/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd__nom.tlef
    fd_io_lef=${PDK_ROOT}/${PDK}/libs.ref/sky130_fd_io/lef/sky130_fd_io.lef
    ef_io_lef=${PDK_ROOT}/${PDK}/libs.ref/sky130_fd_io/lef/sky130_ef_io.lef
    rcx_rules_file=${PDK_ROOT}/${PDK}/libs.tech/sky130_fd_sc_hd/openlane/rules.openrcx.sky130A.nom.calibre
    macros_lef="[glob ./lef/*.lef]"
fi

cmd="
read_lef $tech_lef
read_lef $fd_cells_lef
read_lef $ef_cells_lef
read_lef $fd_io_lef
read_lef $ef_io_lef
foreach lef "$macros_lef" { read_lef \$lef; }
read_def $1
define_process_corner -ext_model_index 0 X
extract_parasitics \
    -ext_model_file $rcx_rules_file \
    -lef_res
write_spef $1.spef
"

openroad -no_init -no_splash <<EOF
$cmd
EOF
```
error:
```
openroad> read_def ./def/gpio_signal_buffering.def
[INFO ODB-0127] Reading DEF file: ./def/gpio_signal_buffering.def
[INFO ODB-0128] Design: gpio_signal_buffering
[WARNING ODB-0293] TECHNOLOGY is ignored
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_28/X, currently on layer li1 at (3381775, 3615625), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_29/X, currently on layer li1 at (3384835, 3618855), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_41/X, currently on layer li1 at (3381775, 3627585), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_59/A, currently on layer li1 at (752525, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_37/X, currently on layer li1 at (3384835, 2236775), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_171/X, currently on layer li1 at (206220, 3010255), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_69/A, currently on layer li1 at (722625, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_58/A, currently on layer li1 at (747895, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_26/X, currently on layer li1 at (3381775, 3609645), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_183/X, currently on layer li1 at (206095, 4452365), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_93/A, currently on layer li1 at (2165095, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_79/A, currently on layer li1 at (688095, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_68/A, currently on layer li1 at (717995, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_57/A, currently on layer li1 at (758505, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_169/X, currently on layer li1 at (206220, 3016235), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_179/X, currently on layer li1 at (206220, 2986335), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_67/A, currently on layer li1 at (723975, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_0/X, currently on layer li1 at (206280, 1752250), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_66/A, currently on layer li1 at (728605, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_55/A, currently on layer li1 at (764485, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_32/X, currently on layer li1 at (3381775, 3633565), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_33/X, currently on layer li1 at (3384835, 3636795), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_159/X, currently on layer li1 at (206220, 3046135), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_177/X, currently on layer li1 at (206220, 2992315), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_186/X, currently on layer li1 at (206095, 4428445), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_95/A, currently on layer li1 at (2171075, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_76/A, currently on layer li1 at (694075, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_65/A, currently on layer li1 at (734585, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_54/A, currently on layer li1 at (759855, 212875), cannot determine exit layer of path
[... 16 lines cut ...]
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_189/X, currently on layer li1 at (206095, 4440405), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_97/A, currently on layer li1 at (2183035, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_72/A, currently on layer li1 at (706035, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_61/A, currently on layer li1 at (746545, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_43/A, currently on layer li1 at (847765, 4984885), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_89/A, currently on layer li1 at (2153135, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_82/A, currently on layer li1 at (676135, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_71/A, currently on layer li1 at (716645, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_60/A, currently on layer li1 at (741915, 212875), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_850_260_hv in net sky130_fd_sc_hd__buf_8_181/X, currently on layer li1 at (206095, 4446385), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_70/A, currently on layer li1 at (712015, 212875), cannot determine exit layer of path
[... cut ...]
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_3/A, currently on layer li1 at (770465, 215935), cannot determine exit layer of path
[WARNING ODB-0110] error: invalid VIA layers in via_1090_260_hv in net sky130_fd_sc_hd__buf_8_1/A, currently on layer li1 at (674785, 215935), cannot determine exit layer of path
[WARNING ODB-0149] DEF parser returns an error!

```
this was run after removing the blockages section in `gpio_signal_buffering` which cause a syntax error.

kareefardi, 2022-11-10:
magic version `8.3.337` fixes the invalid via layers.

mo-hosni, 2023-06-04:
`gpio_signal_buffering` is not being used anymore.

closed 2023-06-04 by mo-hosni

DavidRLindley, 2024-12-10:
I see above that "magic version 8.3.337 fixes the invalid via layers" and "gpio_signal_buffering is not being used anymore."  I'm going to close this ticket because of that and the fact this hasn't been addressed in 18 months.  Please reopen if it is still an issue.