efabless/caravel #375: generation of mag to gds back to mag

marwaneltoukhy, 2022-11-11:
To fix antenna violations on caravel top level, we generated gds out of mag and did the fix on gds then when regenerating the mag the defaults blocks were randomly assigned, the theory is that the blocks lost their property names when generating gds from mag, therefore magic randomly assigned them. The ECO was done on the final mag by returning the correct locations of the default blocks, these locations were copied from the original mag generated by openlane. 

This issue is related to [issue #196 in magic](https://github.com/RTimothyEdwards/magic/issues/196)

RTimothyEdwards, 2022-11-14:
This has been fixed in magic version 8.3.340.

DavidRLindley, 2024-12-10:
Issue was fixed in magic 8.3.340.

closed 2024-12-10 by DavidRLindley