efabless/caravel #393: Broken link descriping power configurations

xobs, 2022-11-23:
The documentation at https://caravel-harness.readthedocs.io/en/latest/caravel-with-openlane.html#option-1-inserting-your-design-macro-into-the-wrapper has a link to a page describing how to configure things for a special power configuration. This link goes to https://openlane.readthedocs.io/en/latest/docs/source/hardening_macros.html#power-grid-pdn which does not exist.

xobs, 2022-11-23:
Presumably the link was meant to go to https://openlane.readthedocs.io/en/latest/usage/advanced_power_grid_control.html or to https://openlane.readthedocs.io/en/latest/usage/hardening_macros.html#power-grid-power-distribution-network

xobs, 2022-11-23:
Actually, most of the links in https://caravel-harness.readthedocs.io/en/latest/caravel-with-openlane.html#extra-pointers are dead, as is a link to https://github.com/efabless/caravel_user_project/blob/main/openlane/user_project_wrapper/config.tcl

DavidRLindley, 2024-12-10:
@algofoogle can you take a look at this?  The links are still broken.  Easy fix?  Do we even have edit permissions there?