efabless/caravel #397: Fill script issue

egorxe, 2022-12-06:
During tapeout of our project https://platform.efabless.com/projects/1184 on MPW-7 rerun we've encountered an issue with fill script. Design passes precheck, but slightly fails M2&M5 density checks **after** metal fill during tapeout (ab7b13dd-56a8-4e15-89da-5b9ca2b410cf). Resulting failing densities are:
m2_ca_density: 0.6592
m5_ca_density: 0.5514

After looking at resulting OASIS file in Open Galaxy I think there are a couple of underfill problems:

1. Square 700x700um pattern of 4um wide unfilled areas.
M2 (layers 69/20+41/28):
[image]
M5 (layers 72/20+59/28):
[image]
This is most likely related to magic box size [here](https://github.com/efabless/caravel/blob/38492d9da4071eef2b9a32029fa9d4778a698cc1/scripts/generate_fill.py#L174).

2. For some reason script does not always put small fill shapes between parallel wires. For M5 this could lead to long empty areas more than 10um in width.
M2:
[image]
M5 (this is the long empty area seen at lower right of previous M5 shot):
[image]

Also I would like to note that this GDS passed tapeouts on MPW6 and first MPW7 run. @d-m-bailey suggested on slack that it may be related to addition of more layers to density calculation in density rules 2 months ago.

whutddk, 2022-12-08:
My new project in MPW-8 https://platform.efabless.com/shuttles/MPW-8/project/1567 may have the same issue. M2 got a violation. And can not pass the tape-out job. Any suggestion?

[image]

DavidRLindley, 2025-01-02:
Thank you for your effort on this.  I completed the 2024 ChipIgnite tapeouts and don't recall having this issue.  I'm closing this ticket due to that and due to the ago of the ticket.   Please reopen the ticket if needed.

closed 2025-01-02 by DavidRLindley