efabless/caravel #400: CRITICAL: Missing metal4 and metal5 routing layers in user_project_wrapper

d-m-bailey, 2022-12-18:
https://open-source-silicon.slack.com/archives/C017HPHCMEY/p1671234224478279
As reported in the slack thread above, metal4 and metal5 routing layers are not included in the final gds.

The result data is here https://drive.google.com/file/d/174xo3M_PDSlZFlpRTHE1TZrOVEAMrfsh/view?usp=share_link

Looking at the gds data in the results/signoff directory
```
2710013505 141440934 ./results/signoff/user_project_wrapper.klayout.gds
2545291106 174227240 ./results/signoff/user_project_wrapper.magic.gds
2710013505 141440934 ./results/signoff/user_project_wrapper.gds
2710013505 141440934 ./results/final/gds/user_project_wrapper.gds
```
you can see that the klayout data (which is the final data) is significantly smaller than the magic data.
Visual inspection of the gds data with klayout show that `user_project_wrapper.magic.gds` has the metal4/5 routing layers (and thus passes LVS) but the `user_project_wrapper.klayout.gds` and `user_project_wrapper.gds` data (same cksum) does not.

There is an XOR check between 2 gds files, but the report header shows that 
`user_project_wrapper.gds` is being compared to `user_project_wrapper.klayout.gds` rather than 
`user_project_wrapper.magic.gds` being compared to `user_project_wrapper.klayout.gds`.

`reports/signoff/user_project_wrapper.xor.xml`
```
<?xml version="1.0" encoding="utf-8"?>
<report-database>
 <description>XOR .../user_project_wrapper/runs/22_12_17_16_07/results/signoff/user_project_wrapper.gds vs. .../user_project_wrapper/runs/22_12_17_16_07/results/signoff/user_project_wrapper.klayout.gds</description>
```

d-m-bailey, 2022-12-18:
@shalan @donn 
Actually this looks like a problem with the `flow.tcl` and related scripts in openlane.
scripts/tcl_commands/klayout.tcl has the  following routine:
```
proc run_klayout_gds_xor {args} {
	set options {
		{-layout1 optional}
		{-layout2 optional}
		{-output_xml optional}
		{-output_gds optional}
	}
	parse_key_args "run_klayout_gds_xor" args arg_values $options
	set_if_unset arg_values(-layout1) $::env(signoff_results)/$::env(DESIGN_NAME).gds
	set_if_unset arg_values(-layout2) $::env(signoff_results)/$::env(DESIGN_NAME).klayout.gds
```
The default layout1 is the output gds. 
This would work is the output gds was from magic, but it looks like the output gds is now from klayout. 

Suggest making the default layout1 to be explicitly magic. That way you can change the output gds any time you want.

kareefardi, 2022-12-18:
The problem is using klayout as a signoff tool in the configuration given in the slack thread. Klayout is not working and should be disabled until it gets fixed.

DavidRLindley, 2025-01-02:
Hi @d-m-bailey Is this still an issue?