efabless/caravel #401: Documentation: link to `info.yaml` is dead

xobs, 2022-12-28:
There is a link at https://caravel-harness.readthedocs.io/en/latest/getting-started.html#required-directory-structure that goes to an example `info.yaml` file at https://github.com/efabless/caravel/blob/main/info.yaml. However, the file does not exist.

mithro, 2022-12-28:
For MPW-1, MPW-2 and MPW-3, all the projects have an info.yml file, see examples below;
 * https://foss-eda-tools.googlesource.com/third_party/shuttle/sky130/mpw-001/slot-001/+/refs/heads/main/info.yaml
 * https://foss-eda-tools.googlesource.com/third_party/shuttle/sky130/mpw-002/slot-001/+/refs/heads/main/info.yaml

However, these files seem to be missing from the MPW-5, MPW-6 and MPW-7 projects? Did something replace these files?

mithro, 2022-12-28:
Somewhat related -> https://github.com/mattvenn/efabless_project_tool/issues/8

mithro, 2022-12-28:
I wrote a quick script at https://github.com/mithro/mpw-tools/blob/main/get-mpw-info-yml.py to download all the `info.yml` files and merge them into a single `.json` file.

proppy, 2022-12-29:
Related https://github.com/efabless/caravel_user_project/issues/230