efabless/caravel #402: fails to simulate with iverilog 13

mattvenn, 2023-01-04:
lots of errors like these:

```
verilog -Ttyp -DFUNCTIONAL -DSIM -DUSE_POWER_PINS -DUNIT_DELAY=#1 \                                                                                                                          
-f/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/mgmt_core_wrapper/verilog/includes/includes.rtl.caravel \                      
-f/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/verilog/includes/includes.rtl.caravel_user_project -o sim_build/sim.vvp rgb_mixer_tb.v               
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:110: error: 'user1_vcc_powergood' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:96:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:111: error: 'user2_vcc_powergood' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:97:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:112: error: 'user1_vdd_powergood' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:98:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:113: error: 'user2_vdd_powergood' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/mgmt_protect.v:99:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:134: error: 'pad_gpio_holdover' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:88:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:135: error: 'pad_gpio_slow_sel' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:89:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:136: error: 'pad_gpio_vtrip_sel' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:90:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:137: error: 'pad_gpio_inenb' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:91:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:138: error: 'pad_gpio_ib_mode_sel' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:92:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:139: error: 'pad_gpio_ana_en' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:93:      : It was declared here as a net.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:140: error: 'pad_gpio_ana_sel' has already been declared in this scope.
/home/matt/work/asic-workshop/shuttle8/zero_to_asic_mpw8/caravel/verilog/rtl/gpio_control_block.v:94:      : It was declared here as a net.
```

mattvenn, 2023-01-04:
seems that the old style of port definition is no longer supported

```    output        user1_vcc_powergood,
    output        user2_vcc_powergood,
    output        user1_vdd_powergood,
    output        user2_vdd_powergood
);

        wire [462:0] mprj_logic1;
        wire         mprj2_logic1;

        wire mprj_vdd_logic1_h;
        wire mprj2_vdd_logic1_h;
        wire mprj_vdd_logic1;
        wire mprj2_vdd_logic1;

        wire user1_vcc_powergood;
        wire user2_vcc_powergood;
        wire user1_vdd_powergood;
        wire user2_vdd_powergood;
```

RTimothyEdwards, 2023-01-04:
@mattvenn :  I expect it is supported, but it may be that it now requires a specific switch.  Check the `iverilog` documentation for command-line options for supported verilog styles (also check if there are any "release notes" with iverilog 13 that would suggest that such a change was implemented).

antonblanchard, 2023-01-05:
I recently hit this, and bisected it to https://github.com/steveicarus/iverilog/commit/6204b78610fedbe36b73fd26bf144f2f07849fb3

The commit mentions:
```
(System)Verilog allows to declare the port direction separate from the
signal declaration. E.g.

output x;
integer x;

But this is only allowed if the port declaration
 * does not have an explicit net type
 * does not have an explicit data type
 * is a non-ANSI style declaration

For all other cases of port declarations the signal is considered fully
defined and it is not allowed to have a separate signal declaration.
```

Assuming this falls under the non-ANSI style declaration, then it appears to be an Icarus Verilog issue.

antonblanchard, 2023-01-05:
I submitted a caravel fix: https://github.com/efabless/caravel/pull/403