efabless/caravel #407: update mask for RDL to not apply polyimide on test tiles

jeffdi, 2023-01-16:


RTimothyEdwards, 2023-01-16:
What is the point of this issue?

jeffdi, 2023-01-16:
Tim A wants test tiles from the Open MPW shuttles without polyimide.  It is
interfering with the processing he is doing with them.

-- Jeff

RTimothyEdwards, 2023-01-16:
If Tim Ansell is paying for another RDL mask set, there are additional changes I would like to make, mainly to add more power and ground pads, and additional pad locations which could be used for future revisions;  these would not change the bump bonding but should be carried up to the RDL level.  (If we can change the bump bonding arrangement, I would add another column of solder bumps and increase the number of available user GPIO pads.)