efabless/caravel #421: caravel-redesign-2: Missing mag/mgmt_core_wrapper.mag

d-m-bailey, 2023-03-07:
The `caravel-redesign-2` branch is missing `mag/mgmt_core_wrapper.mag`. 
The actual layout in this file is irrelevant because `make ship/truck` adds a `GDS_FILE` property.

passant5, 2023-03-07:
the `mgmt_core_wrapper` is flattened inside the `caravel_core`

passant5, 2023-03-07:
@d-m-bailey The only required macro to be loaded from the [caravel_mgmt_soc_litex](https://github.com/efabless/caravel_mgmt_soc_litex/tree/caravel-core) repo is the `RAM128`. should its views be added to the `caravel-redesign-2` branch?

d-m-bailey, 2023-03-07:
The `make ship/truck` target in the `Makefile` loads `mgmt_core_wrapper`.
```
        @echo "\
                random seed `$(CARAVEL_ROOT)/scripts/set_user_id.py -report`; \
                drc off; \
                crashbackups stop; \
                addpath hexdigits; \
                addpath $(CARAVEL_ROOT)/mag; \
                addpath $(UPRJ_ROOT)/mag; \
                path search; \
                load user_project_wrapper; \
                property LEFview true; \
                property GDS_FILE $(UPRJ_ROOT)/gds/user_project_wrapper.gds; \
                property GDS_START 0; \
                load mgmt_core_wrapper; \
                property LEFview true; \
                property GDS_FILE $(MCW_ROOT)/gds/mgmt_core_wrapper.gds; \
                property GDS_START 0; \
                load $(UPRJ_ROOT)/mag/user_id_programming; \
                load $(UPRJ_ROOT)/mag/user_id_textblock; \
                load ../maglef/simple_por; \
                load $(UPRJ_ROOT)/mag/caravel -dereference; \
                select top cell; \
                expand; \
                cif *hier write disable; \
                cif *array write disable; \
                gds write $(UPRJ_ROOT)/gds/caravel.gds; \
                quit -noprompt;" > $(UPRJ_ROOT)/mag/mag2gds_caravel.tcl
```
and sets the `GDS_FILE` property. 

So with `mgmt_core_wrapper` flattened, `caravel_core.mag` directly loads `mgmt_core_wrapper`s subcells `RAM128`, etc. However, `RAM128.mag` is not on the path, so magic throws an error.
```
File ../mag/RAM128.mag couldn't be read
```

Do you want to change the `Makefile` to load `caravel_core` instead of `mgmt_core_wrapper` and set a `GDS_FILE` for `caravel_core` instead?

d-m-bailey, 2023-03-07:
@passant5 I'm not even going to try suggest a simple solution to this because it may be a much deeper issue.

It seems to me that previously hard macro `mgmt_core_wrapper` is now being used as a soft macro where only the verilog from the repo is relevant. Maybe the hard sub macro `RAM128` is also needed from the repo.

There's also this message in regards to the `simple_por`.
```
Duplicate cell in caravel_core:  Instance of cell simple_por is from path ../mag but cell was previously read from ../maglef.
Cell name conflict:  Renaming original cell to simple_por#0.
Warning:  Renaming read-only cell "simple_por"
Read-only status will be revoked and GDS file pointer removed.
```
The `GDS file pointer removed` is alarming. 

@RTimothyEdwards help. Who do we need to bring into this?

marwaneltoukhy, 2023-03-07:
@d-m-bailey `make ship` is not updated. The hierarchy changed quit a bit, and this target is only used for tapeout, so it is usually the last thing that gets updated after we review the hierarchy. This hierarchy is similar to the one in `gf180mcu` so I recommend you take a look at the target [here](https://github.com/efabless/caravel-gf180mcu/blob/3a6527a54c6321076ffb3865dd2241e251365ffe/Makefile#L91).

RTimothyEdwards, 2023-03-07:
@marwaneltoukhy :  Mitch is trying to run LVS on a full chip design so it is critical that `make ship` should keep tracking the actual layout changes.  It is not some detail to be left until the end.

RTimothyEdwards, 2023-03-07:
@d-m-bailey :  The purpose of the sleight-of-hand around the simple POR is that the top level assembly uses `cif *hier write disable` to avoid the considerably large overhead to resolve hierarchical issues in the GDS that aren't needed for properly designed standard cells and I/O cells, all of which correctly abut without modifications.  But that gets disabled universally, so any circuits that require hierarchical processing, like the simple_por, may end up with DRC errors due to the failure to patch up layers through the cell hierarchy.  So the simple_por is (supposed to be) using the abstract view, which will get replaced verbatim with the pre-built simple_por.gds, which has already been put through hierarchical processing.

If the view is switched from "maglef" to "mag", then there is some risk of DRC errors being introduced into the design due to the failure to do hierarchical processing.

Maybe I should implement a different behavior in magic;  I could use a property tag to force a cell to use hierarchical GDS processing;  essentially override `cif *hier write disable` on a local level.  That would avoid needing to deal with the abstract view.

d-m-bailey, 2023-03-07:
@RTimothyEdwards Thanks for the explanation. Just wanted to be sure that people are aware the current `make ship/truck` won't work with `caravel-redesign-2`. I'll try again once the `Makefile` is updated.

marwaneltoukhy, 2023-03-07:
I can update it, but I'll need @jeffdi and @RTimothyEdwards approval and review as the tapeout process heavily depends on it.

marwaneltoukhy, 2023-03-09:
@d-m-bailey fixed in PR #422

closed 2023-06-04 by mo-hosni