efabless/caravel #423: Missing subcut for user_project_wrapper

marwaneltoukhy, 2023-03-14:
When running LVS on full caravel + non empty user_project_wrapper, I get a net mismatch due to `vssd` short, which is fixed when I add a subcut around the user_project_wrapper.

CC: @d-m-bailey @RTimothyEdwards

d-m-bailey, 2023-03-18:
@marwaneltoukhy I'm thinking that we might want to include subcut on the default `user_project_wrapper`/`user_analog_project_wrapper` but allow the users to modify it. @RTimothyEdwards is there a way to programmatically add subckt in openlane before gds out? Maybe add a `SUBCUT_REGION` variable to the openlane config file.

Otherwise, if we use the default for automatically generated layout and they need to modify subcut, they'd need to do post processing - which probably isn't ideal, but I don't see a way around it. Probably change the gds directly in klayout because rewriting the gds from a modified mag file with magic has the potential to create unexpected data if the pdk/search path are different from when the gds was originally created.

RTimothyEdwards, 2023-03-18:
@d-m-bailey :  The ideal solution is that the user puts isosub under every region of the user project that is not the default substrate.  If the user used only the vccd1/vssd1 domain then there would be one isosub region covering the entire user project area.  If they used more than one domain, then they would need to put non-touching isosub regions under each one, which is something we'd have to instruct them how to do.  I wonder if it wouldn't work better to declare `vssd1` to be the default substrate domain, and then we drop isosub everywhere else (but does that work with the I/O pads?).  But I think your idea works, to programmatically add the isosub.  If the user can assume that the default substrate in the user project area is `vssd1`, then we can (with a cifoutput style recipe in magic) draw isosub under the whole thing while avoiding (with a small boundary) any isosub areas the user might have drawn to separate out vssa1, vssa2, or vssd2.

It's kind of involved to think up the solution on the fly, but yes, it can be done.