efabless/caravel #426: make ship fails with certain user ids

d-m-bailey, 2023-03-18:
```
File /home/user/mpw-9/caravel_user_project3/mag/hexdigits/alpha_C.mag couldn't be read
Cell alpha_C is unavailable.  It could not be expanded.
File /home/user/mpw-9/caravel_user_project3/mag/hexdigits/alpha_D.mag couldn't be read
Cell alpha_D is unavailable.  It could not be expanded.
File /home/user/mpw-9/caravel_user_project3/mag/hexdigits/alpha_E.mag couldn't be read
Cell alpha_E is unavailable.  It could not be expanded.
File /home/user/mpw-9/caravel_user_project3/mag/hexdigits/alpha_F.mag couldn't be read
Cell alpha_F is unavailable.  It could not be expanded.
```

d-m-bailey, 2023-03-18:
Looks like it's the last 2 digits of a 8 unique digit string.
Using `01234567`, I get an error with `6` and `7`.

RTimothyEdwards, 2023-03-18:
Isn't it incorrect to be looking in the user project directory for the `hexdigits` subdirectory from caravel?  That should be pointing to the base caravel repository.

d-m-bailey, 2023-03-18:
I believe it's read with `-dereference` and the first 6 unique digits work.

Here's the build command 
```
random seed 1985229328;
drc off; 
crashbackups stop;
addpath hexdigits; 
addpath /home/kanobailey/mpw-9/caravel_user_project3/caravel/mag;
addpath /home/kanobailey/mpw-9/caravel_user_project3/mag;
addpath /home/kanobailey/mpw-9/caravel_user_project3/mgmt_core_wrapper/mag;
load user_project_wrapper; 
property LEFview true; 
property GDS_FILE /home/kanobailey/mpw-9/caravel_user_project3/gds/user_project_wrapper.gds;
property GDS_START 0; 
load /home/kanobailey/mpw-9/caravel_user_project3/mag/user_id_programming;
load /home/kanobailey/mpw-9/caravel_user_project3/mag/user_id_textblock;
load /home/kanobailey/mpw-9/caravel_user_project3/caravel/maglef/simple_por;
load /home/kanobailey/mpw-9/caravel_user_project3/mag/caravel -dereference;
select top cell; 
expand;
cif *hier write disable; 
cif *array write disable; 
gds write /home/kanobailey/mpw-9/caravel_user_project3/gds/caravel.gds;
quit -noprompt;
```
and build log (after deleting `rects` counts and `Scaled` messages).
```
Magic 8.3 revision 373 - Compiled on Wed Mar  1 20:23:37 PST 2023.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Input style sky130(): scaleFactor=2, multiplier=2
The following types are not handled by extraction and will be treated as non-electrical types:
    ubm
Loading sky130A Device Generator Menu ...
Loading "/home/user/mpw-9/caravel_user_project3/mag/mag2gds_caravel.tcl" from command line. 
Processing timestamp mismatches: user_id_textblockWarning:  Parent cell lists instance of "chip_io" at bad file path /home/user/mpw-9/caravel_user_project3/mag/chip_io.mag.
The cell exists in the search paths at ../mag/chip_io.mag.
The discovered version will be used.
, chip_ioWarning:  Parent cell lists instance of "open_source" at bad file path /home/user/mpw-9/caravel_user_project3/mag/open_source.mag.
The cell exists in the search paths at ../mag/open_source.mag.
The discovered version will be used.
, open_sourceWarning:  Parent cell lists instance of "copyright_block" at bad file path /home/user/mpw-9/caravel_user_project3/mag/copyright_block.mag.
The cell exists in the search paths at ../mag/copyright_block.mag.
The discovered version will be used.
, copyright_blockWarning:  Parent cell lists instance of "caravel_core" at bad file path /home/user/mpw-9/caravel_user_project3/mag/caravel_core.mag.
The cell exists in the search paths at ../mag/caravel_core.mag.
The discovered version will be used.
Duplicate cell in caravel_core:  Instance of cell simple_por is from path ../mag but cell was previously read from /home/user/mpw-9/caravel_user_project3/caravel/maglef.
Cell name conflict:  Renaming original cell to simple_por#0.
Warning:  Renaming read-only cell "simple_por"
Read-only status will be revoked and GDS file pointer removed.
, caravel_core, spare_logic_block, sky130_fd_sc_hd__conb_1, sky130_fd_sc_hd__tapvpwrvgnd_1, sky130_fd_sc_hd__decap_3, sky130_fd_sc_hd__fill_2, sky130_fd_sc_hd__decap_8, sky130_fd_sc_hd__decap_4, sky130_fd_sc_hd__fill_1, sky130_fd_sc_hd__decap_6Warning:  Parent cell lists instance of "RAM128" at bad file path ../mag/RAM128.mag.
The cell exists in the search paths at /home/user/mpw-9/caravel_user_project3/mgmt_core_wrapper/mag/RAM128.mag.
The discovered version will be used.
, caravel_clockingWarning:  Parent cell lists instance of "caravel_motto" at bad file path /home/user/mpw-9/caravel_user_project3/mag/caravel_motto.mag.
The cell exists in the search paths at ../mag/caravel_motto.mag.
The discovered version will be used.
, caravel_mottoWarning:  Parent cell lists instance of "caravel_logo" at bad file path /home/user/mpw-9/caravel_user_project3/mag/caravel_logo.mag.
The cell exists in the search paths at ../mag/caravel_logo.mag.
The discovered version will be used.
, caravel_logo.
Warning:  Dereferenced cell "sky130_fd_pr__res_xhigh_po_0p69_S5N9F3" not found in search paths;  using original location ../mag/primitives/sky130_fd_pr__res_xhigh_po_0p69_S5N9F3.
Warning:  Dereferenced cell "sky130_fd_pr__pfet_g5v0d10v5_ZEUEFZ" not found in search paths;  using original location ../mag/primitives/sky130_fd_pr__pfet_g5v0d10v5_ZEUEFZ.
Warning:  Dereferenced cell "sky130_fd_pr__pfet_g5v0d10v5_YUHPXE" not found in search paths;  using original location ../mag/primitives/sky130_fd_pr__pfet_g5v0d10v5_YUHPXE.
Warning:  Dereferenced cell "sky130_fd_pr__pfet_g5v0d10v5_YUHPBG" not found in search paths;  using original location ../mag/primitives/sky130_fd_pr__pfet_g5v0d10v5_YUHPBG.
[... 40 lines cut ...]
string FIXED_BBOX 0 0 11344 8338
string GDS_FILE ../gds/simple_por.gds
string GDS_START 0
string LEFclass BLOCK
<< end >>

==> caravel/maglef/simple_por.mag <==
port 1 nsew power bidirectional
rlabel metal4 38 7965 7241 8283 0 vdd3v3
port 2 nsew power bidirectional
<< properties >>
string LEFclass BLOCK
string FIXED_BBOX 0 0 11344 8338
string GDS_FILE ../gds/simple_por.gds
string GDS_START 0
string LEFview TRUE
<< end >>
```
Is the error message in the build log irrelevant?
```
Duplicate cell in caravel_core:  Instance of cell simple_por is from path ../mag but cell was previously read from /home/user/mpw-9/caravel_user_project3/caravel/maglef.
Cell name conflict:  Renaming original cell to simple_por#0.
Warning:  Renaming read-only cell "simple_por"
Read-only status will be revoked and GDS file pointer removed.
```

RTimothyEdwards, 2023-03-18:
@d-m-bailey :  Weird problems with files not being able to be opened usually indicates that magic hit the filesystem ulimit on opened file descriptors.  It holds files open because that's the way Linux does file locking.  In this case, there should not be multiple processes attempting to write the same files, so the solution is to add the command `locking disable` at/near the top of the script, before any `.mag` files are read.  This command was added to scripts in precheck for handling the GF process, because the GF I/O cells alone would overrun the ulimit.  I assume it's just that the command was never added to the scripts in the sky130 version of Caravel.  This has not happened in the past, but presumably can be triggered for any design with a sufficient number of cells in it coming from `.mag` files.

RTimothyEdwards, 2023-03-18:
@d-m-bailey :  Please raise an issue on the github issue tracker for magic about the timestamp warning messages.  It is one of those low-level problems that I know about but is too low priority to get worked on.  The underlying issue is that `stdout` is buffered but `stderr` is flushed after every write, so if stdout and stderr are combined in the output stream then they can get scrambled together.  The "quick and dirty" solution would be just to output one line per cell with a timestamp issue.

d-m-bailey, 2023-03-19:
> @d-m-bailey : Please raise an issue on the github issue tracker for magic about the timestamp warning messages. It is one of those low-level problems that I know about but is too low priority to get [...]

raised.

> @d-m-bailey : Weird problems with files not being able to be opened usually indicates that magic hit the filesystem ulimit on opened file descriptors. It holds files open because that's the way [...]

That's crazy, but makes sense. Adding `lockign disable` fixes the problem, but I couldn't find this change on the github version of `caravel-gf180mcu`. Maybe it's in the production version only.

closed 2023-03-19 by d-m-bailey

RTimothyEdwards, 2023-03-19:
There should be an issue and pull request on the `caravel-gf180mcu` repository issue tracker?  James Stine ran into the same problem recently when trying to do a local "make ship" on the Caravel GF version.  I fixed the script and pushed the correction to a branch called `fix_make_ship_script`.

d-m-bailey, 2023-03-19:
Yes. I see it now that you pointed it out.