efabless/caravel #429: It is not glitch free when clock switching from ext_clk to pll_clk and vice versa.

TonyHo722, 2023-03-24:
@RTimothyEdwards 

I think it is a document issue in below link
https://github.com/efabless/caravel/blob/31e3d9c1063d4ec987760fb34579e123c9196e8d/docs/other/digital_locked_loop.txt

the code is not glitch free.
https://github.com/efabless/caravel/blob/main/verilog/rtl/caravel_clocking.v#L89-L91

```
WARNING:  The management SoC altering its own clock has not yet been tested
as of this writing;  however, the core clock should be guaranteed to be glitch-
free through transistions from external clock to DLL output and vice versa.
```

RTimothyEdwards, 2023-03-24:
Yes, the code is known to not be glitch-free.  I don't think this has been documented properly, but the recommendation is to use the housekeeping SPI to switch between them and then apply a reset to the processor.  The message about it being guaranteed to be glitch-free needs to be removed.

TonyHo722, 2023-03-25:
@RTimothyEdwards 
Thanks for yout comment.

closed 2023-03-25 by TonyHo722