efabless/caravel #464: complete automatic generation of the datasheet and documentation for caravel

jeffdi, 2023-05-24:


jeffdi, 2023-08-16:
https://docs.google.com/spreadsheets/d/13OvT5-t9aC3Mo8Vx9vasWcwWbUWFhVQPJ_L843LKSXg/edit?usp=sharing

https://docs.google.com/spreadsheets/d/1z83BX8_WSSJWp5-A8g1AMxLVgkzsLswua5Pk6CLA7ug/edit?usp=sharing

jeffdi, 2023-08-16:
[Caravel Datasheet - Housekeeping (1).pdf](https://github.com/efabless/caravel/files/12360473/Caravel.Datasheet.-.Housekeeping.1.pdf)

azwefabless, 2024-01-24:
Andy and Jeff to complete manual generation for Caravel and Caravel+DFT and temp range for this iteration. Automatic to be completed before 2406

jeffdi, 2024-02-07:
Link to an interim public document for the register definitions...

https://github.com/efabless/caravel_board/blob/main/firmware/chipignite/_docs/caravel_registers.pdf

https://github.com/efabless/caravel/blob/main/docs/caravel_datasheet_2.pdf