efabless/caravel #470: caravel_openframe_wrapper `gpio_loopback_one*` and `gpio_loopback_zero*` shifted relative to chip_io_openframe

kareefardi, 2023-06-04:
West, North and some of East pins are shifted. See screenshots taken from caravel_openframe_wrapper gds view:
[image]
 
[image]
[image]

RTimothyEdwards, 2023-06-04:
@kareefardi :  I don't see any such offset.  Are you looking at the .mag views of the cells?  Those are the views used for tapeouts.

RTimothyEdwards, 2023-06-04:
@kareefardi :  Okay, with a little investigation, I think I've sorted out the main issue, which is that the problem lies in my openframe_project_example repository and not in caravel, and the issue should never have been reported here because it is not a caravel issue.  Also, the issue appears to be that the DEF file generated for the wrapper and added to the openlane directory in the example project is outdated with respect to the latest version of caravel.  I have updated the openframe_project_example repository with a DEF file that matches the layout in caravel.

RTimothyEdwards, 2023-06-07:
@kareefardi :  Should be fixed by PR #473 once it's merged into main.