efabless/caravel #471: caravel_openframe `master` branch vs `openframe_development` branch

kareefardi, 2023-06-05:
There has been a couple of PRs adding caravel_openframe to master and then there is also `openframe_development` which is different from master. What are the differences between the already merged openframe and the development branch. Will the development branch be merged to master?

shalan, 2023-06-05:
@RTimothyEdwards We have been using the `master` branch. Switching to `openframe_development` (as requested by you over the weekend) is not a possibility now. It is behind `master` and there are conflicts between the two branches. 

@jeffdi Given the above, I don't think the open frame example tape-out is a possibility today.

RTimothyEdwards, 2023-06-05:
Some additional changes were made to the `openframe_development` branch between March 14 and March 30.  I'm not exactly sure why they would conflict, but they do need to be folded back into the main branch carefully.