efabless/caravel #489: Is the PLL/DLL working or not? and where is its documentation?

Shehab-Bahaa, 2023-08-26:
In the block diagram of the management area of the Caravel, it is mentioned that it has a DLL (https://caravel-mgmt-soc-litex.readthedocs.io/en/latest/_images/block_diagram.png) while in other parts of the documentation, it is usually mentioned that a PLL is used. Take for example (https://caravel-harness.readthedocs.io/en/latest/external-clock.html) and (https://caravel-harness.readthedocs.io/en/latest/gpio.html).

My questions are:
1) Is there a PLL or a DLL?
2) What is the clock frequency entered into the chip from the PCB?
3) How can we use the PLL/DLL to run the block in the user area at a higher frequency than that of the external clock?
4) Would you please direct me to the documentation of them? as I can't find these values here (https://caravel-mgmt-soc-litex.readthedocs.io/en/latest/index.html#features) or here (https://caravel-harness.readthedocs.io/en/latest/index.html).

Thanks in advance.

RTimothyEdwards, 2023-08-27:
Hello @Shehab-Bahaa ,

(1) The DLL circuit is my design.  It is a DLL, not a PLL.  It was substituted in the code in a place where there had originally been a PLL in an earlier version of the design (see the Raven processor on the Efabless marketplace---this had an 8x clock multiplier PLL from X-Fab).  Consequently, a lot of the signal names had "pll" in them.  I have made a few efforts to change "pll" to "dll" everywhere, but those efforts were incomplete.

(2) The development board has a 10MHz clock chip on it.  It would be valid to replace that with any equivalent clock up to 40MHz, which is the design limit of the onboard processor.

(3a) The core clock to the user project is the wishbone clock, and since the wishbone clock is the same clock that is driving the processor, it is also limited to 40MHz.  You can immediately up the speed of the processor and the clock to the user project to 40MHz by turning on the DLL, turning off the DLL bypass, letting the DLL run at, say, 80MHz, by specifying a feedback divider of 8 (80MHz / 8 = 10MHz to match the external clock) and an output divider of 2 (80MHz / 2 = 40MHz for the core clock).

(3b) There is a "user clock 2" input to the user project wrapper which is derived from the same DLL through an independent output divider.  So you can run the core processor at a rate < 40MHz and run the user clock 2 at any multiple of that rate, up to about 100 to 120MHz (you can get somewhat higher numbers by increasing the digital supply from 1.8V up to the maximum rating of about 2.0V).

(4a) This page: https://caravel-harness.readthedocs.io/en/latest/housekeeping-spi.html at least has the housekeeping SPI registers and values defining the DLL controls.  However, it predates the wishbone interface to housekeeping, which allows all the housekeeping registers to be modified from within a C program.  

(4b) This code: https://github.com/efabless/caravel/blob/main/verilog/dv/caravel/mgmt_soc/pll/pll.c configures the DLL from a C program running on the processor, and is currently probably the best resource for understanding how to do that.

If you have any other questions, just ask.

RTimothyEdwards, 2023-08-27:
By the way, the DLL has been fully tested on silicon.  The DLL was effectively disabled on early Caravel versions (through open MPW-five) due to timing closure issues on the digital core.  From MPW-six onward the DLL is fully functional.  See the Efabless "Shuttle Status" page---all shuttles listed as having the "new" Caravel design have a fully functional DLL.