efabless/caravel #494: How to get more than 4 power domain using Caravel?

Shehab-Bahaa, 2023-09-07:
Greetings,

In the Caravel SoC, there are four power domains in the user area, 2 analog and 2 digital. 

1) My first question is how can I get more than 4 power domains using Caravel? I have a design that requires at least 6 or 7 power domains.
2) What is the maximum current that can pass through the power pad?
I would appreciate it if you also referred to the PDK document providing this info along with your answer.

Thanks in advance.

RTimothyEdwards, 2023-09-07:
I would suggest using the Caravan design, which seems perfect for your application (assuming you aren't maxing out the 38 GPIO for digital purposes).  Caravan has three pins along the top that are designed for high-current power supplies.  Then there are also eight additional analog pads that can be used for supplies at lower current.  The high-current ones should be able to take a sustained 300mA current, if I recall correctly (I have that value in a spreadsheet which I need to track down).  The analog signal pads will depend on the current density of metal 3 (same spreadsheet).

RTimothyEdwards, 2023-09-07:
This is the spreadsheet (public):  https://docs.google.com/spreadsheets/d/1N9To-xTiA7FLfQ1SNzWKe-wMckFEXVE9WPkPPjYkaxE/edit#gid=1654372372.  The page "Layer Resistances and Capacitances" also has electromigration limits at the bottom.  These are derived based on material and cross section and are not vendor-supplied values, but should be reasonably trustable.

Using the electromigration limits, the standard bare analog pad should be able to take a sustained current of 38mA, and the three large power pads on caravan should be able to take a sustained current of 215mA each (my recollection was a bit off).

The sky130 pads used for the existing named domains (vccd1, vccd2, vdda1, vdda2) should be able to take about 35mA each.  Note that the vdda1/vssa1 domain has two pads each, so the vdda1 domain can take about 70mA.

Shehab-Bahaa, 2023-09-07:
Thank you so much for your detailed answer, but would you please elaborate more on the "Caravan Design"? as when I searched for it, I couldn't find any information.

RTimothyEdwards, 2023-09-07:
The caravan top level design is in the same repository as caravel.  However, for the user project, you instead start with this repository:  https://github.com/efabless/caravel_user_project_analog and follow the instructions there.