efabless/caravel #497: Cocotb scripts have outdated references?

RTimothyEdwards, 2023-09-19:
There are parts of the cocotb scripts that appear to refer to the structure of caravel as it was some versions ago before the core was restructured.  For example, in file `cocotb/interfaces/caravel.py`, the initialization routine defines:

```
self.hk_hdl      = dut.uut.housekeeping
```

But `uut` is the instance name of the top level module (`caravel`), and `housekeeping` is no longer a child of the top level, it is now a child of the SoC core.  So this line can only be correct if written

```
self.hk_hdl    = dut.uut.chip_core.housekeeping
```

It is not clear to me how the cocotb testbenches even work with this error.

@marwaneltoukhy :  Any ideas?

shalan, 2023-09-19:
@M0stafaRady

M0stafaRady, 2023-09-19:
@RTimothyEdwards , Yes they are outdated. I was suppose to remove them from caravel since I moved the whole cocotb running scripts outside caravel into [caravel-sim-infrastructure](https://github.com/efabless/caravel-sim-infrastructure). All the needed documentation can be found inside the repo. You can find caravel tests [here](https://github.com/efabless/caravel_user_project/tree/main/verilog/dv/cocotb) and [here](https://github.com/M0stafaRady/caravel_cocotb_tests/tree/main/verilog/dv/cocotb/). 

I added a [PR](https://github.com/efabless/caravel/pull/498) to remove these old files 

sorry for the the inconvenience.

M0stafaRady, 2023-09-20:
The location for adding the tests are under the user project repo not caravel repo. Until caravel and the soc is combined, I use the same flow that the user should use, to verify caravel.