efabless/caravel #500: Openframe DEF File PR boundary

marwaneltoukhy, 2023-09-22:
The openframe example def file has the PR boundary overlapping the PR boundary of the `chip_io`, which makes the router rout over pins inside the `chip_io` boundary.

RTimothyEdwards, 2023-09-22:
@marwaneltoukhy :  Are you sure it's the PR boundary, or is it just that the DEF file did not contain blockages where the non-pin metal shapes are?  Can you post a screenshot of where the error is?

marwaneltoukhy, 2023-09-24:
From my understanding, openroad doesn't take the obstructions into consideration from the template def file, it only uses it to get the pin coordinates. As openroad doesn't see the whole chip when hardening the user wrapper, it is legal for it to route anywhere in the PR boundary, which is the source of the problem, because the PR boundaries of the `chip_io` and the wrapper overlap it routed over the pins.
[image]

RTimothyEdwards, 2023-09-24:
@marwaneltoukhy :  Then you need to fix openlane.  I'm sure OpenROAD will respect obstructions in the input DEF file.

RTimothyEdwards, 2023-09-24:
@marwaneltoukhy :  There is another solution I could employ, which would be to create another gpiov2 cell wrapper that would be just for the openframe version and would just remove all the unused pins.  Then I could probably keep everything else exactly as it is currently defined and openlane would be fine routing anywhere within the boundary.

closed 2023-10-19 by jeffdi