efabless/caravel #505: openframe's project missing power pins

marwaneltoukhy, 2023-10-01:
openframe user project is missing power pins, that is because it uses only 1 power domain, this won't pass precheck's XOR or consistency checks.

There are several ways this could be fixed:

1. Add them manually in the gds and gl netlist
2. Have a power ring around the user project like we have in `user_project_wrapper`
3. Get OpenLane to add them, this would probably require an interactive script, and some modifications in the template DEF file. @kareefardi can elaborate.

kareefardi, 2023-10-01:
If we resort to an interactive script. I believe the least path resistance option is to use a custom OpenROAD script that would call the following function https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/ppl#place-individual-pin. The power pins in question have multiple shapes per pin hence the referenced method may not work. If that's the case, we can use odb. This is the way to do without any modifications to the template def and OpenLane. The downsides are using an interactive script and we will have duplicate information about these pins preset in the custom script and the template def.

RTimothyEdwards, 2023-10-02:
Why wouldn't you just drop in the openframe_project_wrapper layout instead of generating it from the DEF (or swap out the one created from DEF with the original layout)?

marwaneltoukhy, 2023-10-15:
Added custom step to fix this issue here https://github.com/efabless/Cheetah_V2/pull/39.

**NOTE**: this step will only work with openlane 2.

closed 2023-12-14 by marwaneltoukhy