efabless/caravel #519: include user process for full chip LVS for the April shuttle

jeffdi, 2023-12-20:


d-m-bailey, 2024-02-09:
This requires the automatic addition of subcut at the `caravel_core` or `caravan_core` level. For openframe, I'm not sure where it should go.

jeffdi, 2024-02-28:
1. Mitch created pseudo flow for adding subcut region.  
2. Mitch to add new output style to the tech file.
3. LVS config file modification
4. Modify make ship / truck to include new cell
5. Confirm versions used with production tapeout (see Omla)
6. Offline verify LVS script with changes above
7. Port to current tapeout for production
8. May need update / cleanup to precheck (some files related to CVC / LVS should move to the pdk)?
9. Test and verify

Target to release 1 month prior to shuttle deadline on April 24th

DavidRLindley, 2024-03-13:
Deferring past April.